MaslowCNC/GroundControl #255: Configuration step 7 of 8 - the only exit from here is 'skip'

blurfl, 2017-05-03:
Some sort of 'Finished' button to indicate that I'm satisfied with the measurements and want to move on. ''Skip' seems like it might even undo any changes I've made.

BarbourSmith, 2017-05-03:
It should automatically move to the next page once it's within tolerance, but maybe that's too much control to the software?

blurfl, 2017-05-03:
The text doesn’t say that will happen. Would it say how close the calibration is? (I know, that’s an enhancement :) )
 I can imagine someone wanting to save their work and ‘move on for now’. A choice of either “getting closer, within xx millimeters - run again” or “good, let’s move on” or some such.

BarbourSmith, 2017-05-05:
I've made the text more clear so that now it says "....The process will repeat until both measurements are within .5mm....." to make it more clear what is happening.

I like the idea of a choice of either “getting closer, within xx millimeters - run again” or “good, let’s move on” or some such.

Would those be two buttons?

blurfl, 2017-05-05:
I think so, yes. Run again is already there...

blurfl, 2017-06-21:
Fixed

closed 2017-06-21 by blurfl