MaslowCNC/GroundControl #284: Manual z-axis is broken

blurfl, 2017-05-19:
Using GC 0.72 and Firmware 70 with manual z-axis control, after the first z-axis prompt is answered, the gcode commands seem to be buffered and executed without waiting for further z-axis changes to be made.

The following program demonstrates, prompting and waiting for the first z-axis change, then running all further steps even though the second z-axis prompt is still pending:

G20
G1 Z0.25
G1 X-1 Y-1 F25 
G1 Z-0.1
G1 X1 
G1 Y1
G1 X-1 
G1 Y-1
G1 Z0.25
G1 X0 Y0 
G1 Z0

BarbourSmith, 2017-05-23:
Yes!

Great spot. 

What's going on is that the machine now has lines of gcode held in its internal buffer which it continues to execute.

The "Hold" button will also now take some time before it has any effect because the machine will execute all the lines of code in the buffer.

We may need a command like "Stop" which acts immediately to pause the machine.

jenmsuth, 2017-06-04:
Hey Bar - I am waiting for my automatic z-axis to arrive, but in the meantime I am trying to build the permanent frame by manually adjusting the z-axis. I am running in to the same issue as @blurfl above. Is there a satisfactory work around to this while the issue is being resolved? Or should I wait until my z-axis arrives?

BarbourSmith, 2017-06-05:
Thank you for reminding me. This should be a top priority because it's a necessary feature that's broken. I can't think of a work around right now other than going back two weeks in the firmware and software. I'll see if I can't get it fixed by the end of the day.

BarbourSmith, 2017-06-05:
Should be fixed in #311 and firmware [#238](https://github.com/MaslowCNC/Firmware/pull/238)

jenmsuth, 2017-06-06:
Thanks Bar! I will test it out.

Best,
Jen

blurfl, 2017-06-21:
Fixed in #311 and Firmware #238

closed 2017-06-21 by blurfl