MaslowCNC/GroundControl #285: GC fails to render G3 properly

blurfl, 2017-05-19:
G3 gcode commands are drawn as straight lines between the endpoints. An example can be seen with this program:

G20
G1 Z0.25
G1 Y-2 F25 
G1 Z-0.1
G3 Y2 J2
G3 Y-2 J2 
G1 Z0.25
G1 X0 Y0
G1 Z0

blurfl, 2017-05-19:
GroundControl0.71 works correctly, the issue seems to have come with version 0.72

johnbenweber, 2017-05-20:
Try PR #287.

blurfl, 2017-05-20:
PR #287 does indeed resolve this, Very well done 💯 👍

BarbourSmith, 2017-05-23:
This has to do with the way  lines are rendered. Making each line of gcode it's own object on the canvas was really slowing the program down, now just the points that make up the line are stored as part of one continuous object. This is simple for lines, but for curves it takes a little more work.

BarbourSmith, 2017-05-26:
This should be fixed in #293

blurfl, 2017-05-26:
👍

closed 2017-05-26 by BarbourSmith