MaslowCNC/GroundControl #296: Terminal shows 'Unable to check buffer' error at end of cut

blurfl, 2017-05-27:
I have a moderately curvy file that clearly showed the chattering issue mentioned several days ago in these links:
http://www.maslowcnc.com/forums/#!/general:new-gc-release-72/well-cutting-in-the-week-sp
http://muut.com/u/maslowcnc/s3/:maslowcnc:2g9h:img_0827.jpg.jpg

I downloaded GC and Firmware (ref #295 and #230 'sun-moon...') and ran my file and I can report that the chattering is so much reduced as to be practically indistinguishable. 👍 

At the end of the cut, however, I no longer have control of the sled without quitting and rerunning GC. Looking at the terminal window I see the following message repeating:
Unable to check buffer
[PosError:1.01,0.62,160]

Unable to check buffer
[PosError:0.40,0.36,160]

BarbourSmith, 2017-05-27:
I experienced this also. I will look into it.

I was able to regain control by clicking "Stop" several times. Not ideal.

I'm glad to hear that you found the chattering issue to be much less. It's good to get a second opinion. 👍 👍

BarbourSmith, 2018-01-10:
Thank goodness we've solved this one 😃

closed 2018-01-10 by BarbourSmith