MaslowCNC/GroundControl #297: Loss of position should stop sled movements

GeroBH, 2017-05-27:
If the position is lost, the sled on the screen disappears and shows up somewhere else. The router moves towards that position and the screen sled towards the routers actual position until they meet. Then both move back to the cut line. This all happens with Z down in the material and is a good candidate for breaking bits.
A position loss should clear Z to safety hight first before attempting any moves or recovery.

davidelang, 2017-05-27:
and since it's really impossible to know what the safety hight is, if the 
firmware looses track of where it is, it really needs to stop and have the user 
re-do the chain length calibration

BarbourSmith, 2017-05-27:
I think what we were seeing with the file in the video was that the buffer was overflowing (because I swapped a '+' and a '-' in checking how much space was available in the buffer, *face palm*). Once the buffer overflowed the machine would try to execute some junk gcode and end up with a command to be in a different location. I'm not sure it was actually loosing track of where it was, it might have been loosing track of where it should be. That being said, from the video it really did look like it was loosing track of both.

What are your thoughts @GeroBH? Did the fix fix the issue?

GeroBH, 2017-05-28:
The 6 circles went through without any buffer overflow. I had another issue with the right chain getting to long on the outside square,  but suspect the screw on the right sprocket might be loose. Will check tonight.

GeroBH, 2017-05-28:
I can confirm that the fix solved the issue. Run it twice and going for the final cut now. The title however is not the buffer overflow, rather the behaviour after losing position, with the bit in the material. If you don't mind I will keep this open until I see a  'Motors Stop' and a 'Z-Up' before any recovery attempts on the position take place.

davidelang, 2017-05-28:
The problem with wanting a "Z up" when things go wrong is that we don't have a 
limit switch for the Z axis movement, so if you are too near the upper limits of 
your travel, you will strip out the orange plastic nut that the r2200 uses. If 
we did have a limit switch, I would be fully in favor of backing out until you 
hit that limit switch.

But since we don't, I'll argue that there is nothing the machine can do safely 
other than panic and tell the user that they need to fix the position manually.

BarbourSmith, 2017-05-28:
The other issue is that I'm not sure the machine knew that something had gone wrong. After the buffer overflowed the firmware was reading multiple lines of gcode merged into one. Somehow when those lines were merged together it resulted in a command to move suddenly to somewhere else. 

Maybe the answer is to put some logic in there that prevents the machine from ever trying to teleport to a different part of the sheet.

GeroBH, 2017-05-28:
Ok, the final cut went well (pic on the corner case thread). I will close this one, as a have no answer to the question if we can catch buffer overflows and position loss and place a routine for that.
I still see plenty of 'unable to read buffer' and 'unable to plot position on the screen' in the Konsole I start GC from. It doesn't cause any casualties though.
The machine was moving the sled and the false position together until they meet, goes back to the cut and continues exact on the line as if nothing has happened. So there is some routine that brings it back on track. How come it did not know it was lost? So what triggered the recovery then? I wish I had the experience and time to read in to the code.

closed 2017-05-28 by GeroBH