MaslowCNC/GroundControl #300: Error while cutting complex curve.

blurfl, 2017-05-28:
I moved this here from Firmware#234 

Using GC with PR#298 and Firmware from PR#233 to cut a file that amounts to five 12MM diameter pocket cuts 3mm deep. I originally saw the error using metric coordinates so I created a new version using inch values. The error occurred with the inch version as well, though at a slightly differewnt line in the file.
Home has been moved 5" in the Y direction. The error seems to have occurred while executing line 54, the cutter headed west, clearly in error.
Here is the section of the file showing that line:
[image]

Here is the control panel showing the interpretation. The last correct line looks like it was line 53, the following line is incorrect:
[image]

This is the file in use when the screen shots were made:
[Magnet array 12mm x 6cm x 5-SAE.nc.zip](https://github.com/MaslowCNC/GroundControl/files/1034425/Magnet.array.12mm.x.6cm.x.5-SAE.nc.zip)

davidelang, 2017-05-28:
there was a bug in GC that caused buffer overflow errors. This was fixed late 
yesterday, check that you have the latest version, and check to see if you have 
any errors show up while you are cutting.

blurfl, 2017-05-28:
That was PR#298, the version I'm using. First thing I did when the error happened was to check the pertinent file for the change, yes.

BarbourSmith, 2017-06-01:
It's very interesting that just the Y term disappeared completely. That could be a Ground Control issue, but it could also be a firmware thing. Let's keep the issue open here. It's so nicely documented, it would be a shame to move it.

blurfl, 2017-06-01:
On different runs I've seen the z term lost, the decimal point and part of the mantissa lost, and one time where only the letter Y was there. Was there a change to a buffer routine, or are we running low on ram?

davidelang, 2017-06-01:
there have been changes to the buffering in the last couple of weeks. There were 
errors that bar fixed in this week's release, but he may not have got all of 
them.

loosing the tail end of the command does sound like a buffer problem.

BarbourSmith, 2017-06-02:
I've actually noticed the '\n' (an invisible character that marks the end of the line) getting lost pretty regularly too.

I'll start testing by running the file attached here (Thank you!)

If I can't track it down that way, I'll make a firmware GC pair where we just send a known line back and forth as fast as possible and check that it arrives correctly on both ends

davidelang, 2017-06-02:
hmm, if we're loosing \n routinely, that makes me think there may be an 
off-by-one error in there somewhere.

blurfl, 2017-06-02:
I've started looking into Firmware PR#229, the returnError() routine. I wonder if that's too much function calling and printing going on. I've figured out that I can't just comment the print statements out - because the ringBuffer.spaceAvailable() value gets used in serialPortThread.py - but I could shorten things by replacing the previous two function calls and associated print statements with "0,0,". This still has the problem, though, so not the solution.

BarbourSmith, 2017-06-02:
I've got some bad news which is that I'm not seeing the same behavior. I've run the file on three different places on the sheet, and in all three it ran to the end without incident. 

This is going to make it harder for us to figure out what is going on. I would bet that the issue is in some way dependent on the computer itself (the only thing different between our setups). That doesn't mean it's not a problem with our software, just that it's a problem which only pops up under certain circumstances. 

It makes sense to me that we would see this issue now and with this file. The file has a large number of small lines of gcode which are executed very quickly. 

When you have a chance @blurfl give the branch [fix for 300](https://github.com/MaslowCNC/GroundControl/tree/fix-for-300) a try. I've added some print statements to keep an eye on the health of the computers internal serial buffers for both transmitting and receiving.

The **in** number indicates how many bytes of data are waiting to be processed in the incoming buffer and the **out** number indicates how many bytes of data are waiting to be processed on the out going buffer. On my system the out number stays at 0 pretty much all the time because the arduino is pretty good about reading new data right away, while the in hovers under 100.

[image]

What do you see on your end? Is it similar? Do those numbers change at all when you see the error crop up?

blurfl, 2017-06-02:
Good news, actually - it's another data point! I'll try the branch you ask, and if I need to, I'll fire up the RPi and see how it goes on that platform. If need be, I'll drag the Win10 VM out to the shop. 
It'll be later tonight before I can get to it, but I'll let you know what I see. Thanks for giving it your attention!

blurfl, 2017-06-03:
I took a moment to install the 'fix-for-309' version of Firmware and run the 'fix-for-300' version of GC with the .nc file. I put the terminal window next to the GC window and stopped as quickly as I could after the error. I've drawn an arrow between the two to show which line I think is in question. From your description of the serial buffer counters, they look normal, yes?

[image]

blurfl, 2017-06-03:
I really do think this issue is related to the buffering code added to the Firmware in release 72. 
Trying to test this, I've gone back to earlier release pairs. So far I've tried every GC/Firmware release pair since F69/GC71 and my test file works correctly in all except F72/GC74. To test whether the issue was GC74 or the Firmware72, I checked with F71 and GC74 and there was no error. To check for a baud rate problem in F72/GC74 I've tried 19200 baud and themerror occurred. I've done a successful run at 115200 baud using GC74 with F71, but the error crops up if I then install F72. (I've seen a suggestion on a different CNC site that low baud rates actually slow down an Arduino Uno.)
By this I think I've narrowed the issue down to something in F72, and something other than the baud rate. 
I was going to try F72/GC74 on a RPi but I find that my RPi posts python errors with GC74. I haven't taken time to step back to see whether previous releases work on my RPi setup - its been a couple months since I last used it.

davidelang, 2017-06-03:
This is exactly the sort of thing that git bisect was designed for 
https://git-scm.com/docs/git-bisect

blurfl, 2017-06-03:
That's an interesting tool, thanks. Much to learn, CNC is just a part of it 🙂

blurfl, 2017-06-04:
I modified the Firmware to echo each character received out the Serial3 port, which comes out AUX4 on the motor board. Capturing this stream, I see that the error is coming from GC and not the Firmware. The stream seen on Serial3 is the same as that executed and echoed back to GC by the Firmware.
 To check whether the fault is hardware or software, I tested with Linux running on a different laptop and found the same thing, characters missing from the serial stream. I changed cables and saw no difference so the cables are nto the cause. I ran the .nc file using UGC on both platforms and the cut was correct in both cases. When I switch releases back to F71GC73, the cut is correct on both platforms. 
 If I were a better pythonista, I would have GC.serialPortThread.py log the characters to a file as it writes them to the serial port. There is a buffer in GC.serialPortThread.py, perhaps this one is the culprit?

blurfl, 2017-06-04:
I wonder whether we are swamping the pyserial out buffer. Serial.outWaiting() is available to check for buffer space, but that's over my head... A higher baud rate might drain the buffer faster, I'll try that.

davidelang, 2017-06-04:
well, right now we seem to have a problem where Ground Control is not outputting 
what it thinks it is. We need to get logs to show what it thinks it's sending 
out vs what the thing on the other end of the serial port says that it's 
getting.
k

blurfl, 2017-06-04:
OK, I've found the output of the print statement in  GC.serialPortThread.py _write() during an error:

   ding: G1 X146.315685279 Y60.0479695431
   ding: G1 X146.275076142 Y60.1418781726
   ding: G1 X146.216700508 Y60.2230964467
   ding: G1 X146.143096447 Y60.2967005076
   ding: G1 X146.061878173 Y60.3550761421
   ding: G1 X145.967969543 Y60.3956852792

And on the motor board, the serial echo sends:

G1 X146.315685279 Y60.0479695431
G1 X146.275076142 Y60.1418781726
G1 X14G1 X146.143096447 Y60.2967005076
G1 X146.061878173 Y60.3550761421
G1 X145.967969543 Y60.3956852792

The third line seems to have been correct when GC passed it to serialInstance.write but truncated when it arrived at the other end of the wire.

Note: I repeated this test at 230400 baud to see whether draining the buffer faster helped. It did not (though otherwise, the communication seemed to keep up at that rate :) ).

davidelang, 2017-06-04:
can you tweak the software to display how much buffer space it thinks there is 
before each write (and ideally the position in the buffer)?

how repeatable is this error?
k

blurfl, 2017-06-04:
To be more complete, I'll include logs from the last Z movement, because during this part of the cut the GC buffer check triggered:

---------------------------------------------

Lines sent by GC:

   ding: G0 Z5
   ding: G0 X145.324314721 Y60.2241116751
   ding: G1 Z-1.8 F800
   ding: G1 X145.264923858 Y60.1418781726 F1500
   ding: G1 X145.224314721 Y60.0479695431
Run Paused
Unable to check buffer
[PosError:0.302,211]
Run Resumed
   ding: G1 X145.196395939 Y59.9515228426
   ding: G1 X145.188781726 Y59.85
   ding: G1 X145.196395939 Y59.7484771574
   ding: G1 X145.224314721 Y59.6520304569
   ding: G1 X145.264923858 Y59.5581218274
   ding: G1 X145.323299492 Y59.4769035533
   ding: G1 X145.396903553 Y59.4032994924
   ding: G1 X145.478121827 Y59.3449238579
   ding: G1 X145.572030457 Y59.3043147208
   ding: G1 X145.668477157 Y59.2763959391
   ding: G1 X145.77 Y59.2687817259
   ding: G1 X145.871522843 Y59.2763959391
   ding: G1 X145.967969543 Y59.3043147208
   ding: G1 X146.061878173 Y59.3449238579
   ding: G1 X146.143096447 Y59.4032994924
   ding: G1 X146.216700508 Y59.4769035533
   ding: G1 X146.275076142 Y59.5581218274
   ding: G1 X146.315685279 Y59.6520304569
   ding: G1 X146.343604061 Y59.7484771574
   ding: G1 X146.351218274 Y59.85
   ding: G1 X146.343604061 Y59.9515228426
   ding: G1 X146.315685279 Y60.0479695431
   ding: G1 X146.275076142 Y60.1418781726
   ding: G1 X146.216700508 Y60.2230964467
   ding: G1 X146.143096447 Y60.2967005076
   ding: G1 X146.061878173 Y60.3550761421
   ding: G1 X145.967969543 Y60.3956852792
   ding: G1 X145.871522843 Y60.4236040609
   ding: G1 X145.77 Y60.4312182741
Gode Stopped
Sending: 

 ---------------------------------------------
 

 Lines received by motor board serial echo:
  
  
G0 Z5
G0 X145.324314721 Y60.2241116751
G1 Z-1.8 F800
G1 X145.264923858 Y60.1418781726 F1500
G1G1 X145.196395939 Y59.9515228426
G1 X145.188781726 Y59.85
G1 X145.196395939 Y59.7484771574
G1 X145.224314721 Y59.6520304569
G1 X145.264923858 Y59.5581218274
G1 X145.323299492 Y59.4769035533
G1 X145.396903553 Y59.4032994924
G1 X145.478121827 Y59.3449238579
G1 X145.572030457 Y59.3043147208
G1 X145.668477157 Y59.2763959391
G1 X145.77 Y59.2687817259
G1 X145.871522843 Y59.2763959391
G1 X145.967969543 Y59.3043147208
G1 X146.061878173 Y59.3449238579
G1 X146.143096447 Y59.4032994924
G1 X146.216700508 Y59.4769035533
G1 X146.275076142 Y59.5581218274
G1 X146.315685279 Y59.6520304569
G1 X146.343604061 Y59.7484771574
G1 X146.351218274 Y59.85
G1 X146.343604061 Y59.9515228426
G1 X146.315685279 Y60.0479695431
G1 X146.275076142 Y60.1418781726
G1 X14G1 X146.143096447 Y60.2967005076
G1 X146.061878173 Y60.3550761421
G1 X145.967969543 Y60.3956852792
G1 X145.871522843 Y60.4236040609
G1 X145.77 Y60.4312182741
!
   
 ---------------------------------------------

blurfl, 2017-06-04:
Thanks for remindsing me, @BarbourSmith has a [branch ](https://github.com/MaslowCNC/GroundControl/tree/fix-for-300) that does that - I'll capture that as well.

davidelang, 2017-06-04:
Ok, the good news is that this erro correlates to the "unable to check buffer" 
error (which makes sense), so why did that error trigger?

davidelang, 2017-06-05:
hmm, why are we inventing a new protocol rather than just using xon/xoff 
signaling?

we shouldn't have to guess how much space is left in the buffer, we should just 
turn on xon/xoff for the serial connection in python and then implement it on 
the arduino side (and I think it's stupid for the arduino hardware to not 
support hardware flow control :-)

blurfl, 2017-06-05:
Using the 'fix-for-300' branch, these logs:

Lines logged by GC:

   ------------------------------------------

Connected on port /dev/tty.usbmodem14441
in: 80	- out: 	0  : B03 A2438.4 C1219.2 Q2995.65 E369.29 F310.2 R139 H79 I1 J8148.0 K10 M6.35 N3.17 P7560.0   
in: 121	- out: 	0  : B05   
in: 176	- out: 	0  : G21   
in: 26	- out: 	0  :    
in: 36	- out: 	0  :   
    45	- out: 	0  : G21 G90 G40
in: 28	- out: 	0  :   
in: 17	- out: 	0  :   
    48	- out: 	0  : G0 Z5
    47	- out: 	0  : T0 M6
    12	- out: 	0  : G17
    205	- out: 	0  : M3
    77	- out: 	0  : G0 X151.518984772 Y104.223096447
    51	- out: 	0  : G1 Z-1.8 F800
    20	- out: 	0  : G1 X151.422030457 Y104.195685279 F1500
    0	- out: 	0  : G1 X151.328121827 Y104.155076142
    20	- out: 	0  : G1 X151.246903553 Y104.096700508
    66	- out: 	0  : G1 X151.173299492 Y104.023096447
    129	- out: 	0  : G1 X151.114923858 Y103.941878173
    65	- out: 	0  : G1 X151.074314721 Y103.847969543
    95	- out: 	0  : G1 X151.046395939 Y103.751522843
    26	- out: 	0  : G1 X151.038781726 Y103.65
    7	- out: 	0  : G1 X151.046395939 Y103.548477157
    174	- out: 	0  : G1 X151.074314721 Y103.452030457
    167	- out: 	0  : G1 X151.114923858 Y103.358121827
    102	- out: 	0  : G1 X151.173299492 Y103.276903553
    102	- out: 	0  : G1 X151.246903553 Y103.203299492
    87	- out: 	0  : G1 X151.328121827 Y103.144923858
    18	- out: 	0  : G1 X151.422030457 Y103.104314721
    6	- out: 	0  : G1 X151.518477157 Y103.076395939 <---EOL lost
    38	- out: 	0  : G1 X151.62 Y103.068781726
    73	- out: 	0  : G1 X151.721522843 Y103.076395939
    129	- out: 	0  : G1 X151.817969543 Y103.104314721
    112	- out: 	0  : G1 X151.911878173 Y103.144923858
    74	- out: 	0  : G1 X151.993096447 Y103.203299492
    5	- out: 	0  : G1 X152.066700508 Y103.276903553
    19	- out: 	0  : G1 X152.125076142 Y103.358121827
    6	- out: 	0  : G1 X152.165685279 Y103.452030457
    66	- out: 	0  : G1 X152.193604061 Y103.548477157
    21	- out: 	0  : G1 X152.201218274 Y103.65
    31	- out: 	0  : G1 X152.193604061 Y103.751522843
    127	- out: 	0  : G1 X152.165685279 Y103.847969543
    174	- out: 	0  : G1 X152.125076142 Y103.941878173
    181	- out: 	0  : G1 X152.066700508 Y104.023096447
    181	- out: 	0  : G1 X151.993096447 Y104.096700508
    194	- out: 	0  : G1 X151.911878173 Y104.155076142
    125	- out: 	0  : G1 X151.817969543 Y104.195685279
    101	- out: 	0  : G1 X151.721522843 Y104.223604061
    75	- out: 	0  : G1 X151.62 Y104.231218274
    6	- out: 	0  : G1 X151.518477157 Y104.223604061
    19	- out: 	0  : G1 X151.187005076 Y106.106345178 F1500 <--- chars lost between Y1 & EOL
    6	- out: 	0  : G1 X150.767208122 Y105.992639594 F1500
    5	- out: 	0  : G1 X150.373807107 Y105.809898477
    22	- out: 	0  : G1 X150.015939086 Y105.561167513
Gode Stopped
in: 26	- out: 	0  : !  

   ------------------------------------------
 

Lines logged by motor board:

G21 G90 G40
  
  
G0 Z5
T0 M6
G17
M3
G0 X151.518984772 Y104.223096447
G1 Z-1.8 F800
G1 X151.422030457 Y104.195685279 F1500
G1 X151.328121827 Y104.155076142
G1 X151.246903553 Y104.096700508
G1 X151.173299492 Y104.023096447
G1 X151.114923858 Y103.941878173
G1 X151.074314721 Y103.847969543
G1 X151.046395939 Y103.751522843
G1 X151.038781726 Y103.65
G1 X151.046395939 Y103.548477157
G1 X151.074314721 Y103.452030457
G1 X151.114923858 Y103.358121827
G1 X151.173299492 Y103.2769035599492
G1 X151.328121827 Y103.144923858
G1 X151.422030457 Y103.104314721
G1 X151.518477157 Y103.076G1 X151.62 Y103.068781726  <------ error here
G1 X151.721522843 Y103.076395939
G1 X151.817969543 Y103.104314721
G1 X151.911878173 Y103.144923858
G1 X151.993096447 Y103.203299492
G1 X152.066700508 Y103.276903553
G1 X152.125076142 Y103.358121827
G1 X152.165685279 Y103.452030457
G1 X152.193604061 Y103.548477157
G1 X152.201218274 Y103.65
G1 X152.193604061 Y103.751522843
G1 X152.165685279 Y103.847969543
G1 X152.125076142 Y103.941878173
G1 X152.066700508 Y104.023096447
G1 X151.993096447 Y104.096700508
G1 X151.911878173 Y104.155076142
G1 X151.817969543 Y104.195685279
G1 X151.721522843 Y104.223604061
G1 X151.62 Y104.231218274
G1 X151.518477157 Y104.223604061
G1 X151.187005076 Y1                   <-------- error here 
G1 X150.767208122 Y105.992639594 F1500
G1 X150.373807107 Y105.809898477
G1 X150.015939086 Y105.561167513
!

blurfl, 2017-06-05:
Flow control could help, but since it's not part of the library, it's still "roll-your-own".

davidelang, 2017-06-05:
digging into the xon/xoff idea, it looks like the arduino boards are broken enough to not support it https://github.com/grbl/grbl/issues/50

It looks like we need to increase the safety factor and try not to send data up to the full capacity of the receive buffer. the link above talks about sending up to 64 characters in a 'packet' via the usb interface, so what if we were to bump up the size of the receive buffer by 64 characters in the firmware, but leave it set as-is on the GC side? this seems wasteful, but may be what we need to do to avoid this sort of grief.

blurfl, 2017-06-05:
Looking into how they handled comm. in grbl is a good idea.

blurfl, 2017-06-05:
FWIW, I just tried stepping down to 9600baud. No joy.

davidelang, 2017-06-05:
make sense. I don't thin this is a timing problem, I think it's an artifact of 
using USB to send data to simulate a serial port. Serial ports handle characters 
one at a time (with handshaking happening per-character), the USB protocol has a 
lot of overhead per packet, so it sends a bunch of characters at a time.

Trying to have the protocol keep the two sides in sync so that they both know 
the size of the buffer and never send more data than there is room to hold is 
fragile, that's why I suggest making the receive buffer noticably larger so that 
if too much data is sent, it doesn't get lost.
k

blurfl, 2017-06-05:
Just read about upping the Rx buffer on the Arduino [here](https://internetofhomethings.com/homethings/?p=927). I set both Rx and Tx buffers up to 256, no joy.

blurfl, 2017-06-05:
I think we need to throttle the pyserial end, before the USBserial driver's buffer.

blurfl, 2017-06-05:
@BarbourSmith, the missing EOLs that you see are an example of the issue.

BarbourSmith, 2017-06-05:
The good news is we know we (I) introduced this problem a in the last few weeks of big serial changes so we know that the problem can be solved.

I think the first step is to find out exactly which line change caused the issue to first appear. Then we can deduce from that how to fix it.

Which was the last version which worked consistently?

blurfl, 2017-06-05:
I went back to Firmware 70/GC72, and now that I know what to look for, a truncated line like the one that starts G1 X16.76 here:  - 

[image]

This line came in from the serial port as:
G1 X16.7639593909 Y108.34010G1 X16.0 Y108.406091371
Looking at the hex dump of that serial input stream, there are no characters between 'Y108.34010' and 'G1 X16.0'.

Nevertheless, now that I know how to spot it on the GC control panel, I can see evidence of missing characters as far back as  Firmware 70/GC72.

BarbourSmith, 2017-06-05:
That will be a big help in finding what's going on.

If you go back further the problem disappears?

blurfl, 2017-06-05:
Odd thing, that. With Firmware0.69 loaded running GC71, the motor test works, the z axis works, but the right and left motors don't move. Same situation on the Linux box. A puzzler...

BarbourSmith, 2017-06-05:
Hmm, that is a puzzler. 

Could it have something to do with the fact that we've gotten rid of the motor calibration process? Maybe that needs to be run again.

davidelang, 2017-06-05:
The motor calibration process was creating a static power table, the PID loop 
eliminates the need for that. If running the motor calibration process 'fixes' 
anything, it's only because of some unplanned side-effect.

blurfl, 2017-06-05:
Thanks for the reminder. Motor calibration seems to be the answer. I'll run the cut a little later after the calibration has finished.

blurfl, 2017-06-05:
The pair Firmware69/GC71 works without the error. Each of the release pairs after that exhibit the error. Here is a file that causes the error more oftern, probably because of the length of the mantissa for each value :) .
[Magnet Circles Sketchup2.nc.zip](https://github.com/MaslowCNC/GroundControl/files/1053432/Magnet.Circles.Sketchup2.nc.zip)

davidelang, 2017-06-06:
can you try git bisect between that pair and the next version?

commit 3b4e400b62b3073cd34347e8cc9ba38b2ea28d57 is "push lines of gcode until 
buffer is full", which sounds like it's probably the issue, but there were 
several updates to the serial communication in this release.

I suspect this is going to be a case of "yep it broke something" and we will be 
back to troubleshooting exactly why.

This is when the ringbuffer was introduced, allowing multiple commands to be 
sent and buffered, etc.

blurfl, 2017-06-06:
It sound like you're the most experienced with git bisect. Can you do this? Would probably be speedier, yes?

 I'm only a puppy here; I've gotten this far:
git bisect start
git bisect good 61391096330224c7a84fd1e17eac4e6e76c2906c // the SHA of PR bump to version71 
git bisect bad cb0da072eb00a37b7714cf035bbb1ca737bc0f8e // the SHA of PR bump to version72

which tells me:
Bisecting: 8 revisions left to test after this (roughly 3 steps)
[59d09f723dc845a4dbf69d3206eed88a8f03565e] experimenting with points

davidelang, 2017-06-06:
unfortunantly, I have no way to test the result :-(

On Mon, 5 Jun 2017, Scott Smith wrote:

> It sound like you're the most experienced with git bisect. Can you do this? Would probably be speedier, yes?
>
> I'm only a puppy here; I've gotten this far:
> git bisect start
> git bisect good [...]

so what you need to do now is to run this version and then tell git bisect if it 
works or not, at that point git will checkout a different version.

with 8 commits, you could test them one at a time, git bisect will probably find 
the cutoff in 3 steps (the savings are higher the more commits there are)

blurfl, 2017-06-06:
Shoot me .zipped files to run?

davidelang, 2017-06-06:
I had a thought, we are able to use a grbl gcode sender instead of ground 
control to send g-code to the firmware. If you try using that do you have the 
same problem?

If not, that would say it's a GroundControl problem.
If the problem is still there, that would say it's a firmware problem.

This should let us know which side of things we should be looking at.

blurfl, 2017-06-06:
I ran the .nc file using Universal Gcode Sender on both OS X and Ubuntu17.4 and the cut was correct in both cases. 
I have not yet wired up a different USBserial adapter to take the Arduino drivers out of the equation.

davidelang, 2017-06-06:
Ok, that makes it look like it's the GroundControl side sending data when there 
isn't enough room in the buffer.

I will say that _checkBufferSize looks odd to me in serialPortThread.py

it has a couple paths that will set self.machineIsReadyForData to true, but 
nothing that will set it to false. If I am reading it correctly (minutes after I 
wake up, so a bit groggy still), if there isn't room in the buffer, it will 
throw the exception (unable to to check buffer), there is no path for when it 
can check the buffer, but there just isn't room right now

It seems to me that if we are doing the queue management on the GC side, we 
should be doing something along the lines of keeping track of the length of each 
message that we have sent, and when we get an 'ok\n' (but we look for 'ok\r\n') 
we should decrease the amount of buffer used by the length of the oldest message 
and check if we can send a new message, repeating every time we get an ok 
message.

but as I say, I'm not fully awake yet, so I could be missing something here.

On Tue, 6 Jun 2017, Scott Smith wrote:

> Date: Tue, 06 Jun 2017 08:28:09 -0700
> From: Scott Smith <notifications@github.com>
> Reply-To: MaslowCNC/GroundControl
>     <reply+0021b24f321f90956c15ef36de7287058312f8f262f9b49b92cf00000001154e8c0
>     992a169ce0dd25412@reply.github.com>
> To: MaslowCNC/GroundControl <GroundControl@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [MaslowCNC/GroundControl] Error while cutting complex curve.
>     (#300)
> 
> I ran the .nc file using Universal Gcode Sender on both OS X and Ubuntu17.4 and the cut was correct in both cases.
> I have not yet wired up a different USBserial adapter to take the Arduino drivers out of the equation.
>
>

blurfl, 2017-06-06:
valz[2][0:-3] is ringbuffer space available as sent from the Firmware. When I patched serialPortThread.py to print this every time it checks to send, I see it most usually at 256, occasionally lower.

davidelang, 2017-06-06:
right, but that routine is only called when we get a [ back from the firmware, 
the ok is basically ignored, but that is the only time when there should be the 
possibility of more room being available

also remember that we have the error 'unable to check buffer' when we loose 
data, which means that those tests are not passing, but we are still sending 
data.

blurfl, 2017-06-06:
The value is sent by the firmware in CNC_functions.h each time a gcode line is parsed.

davidelang, 2017-06-06:
it doesn't matter when it's set in the firmware, what matters is when it's read 
in GC.

according to the thread about grbl, it tracks when a command is completed by 
looking for the ok, not by getting any buffer size message from the firmware.

trying to get buffer size messages from the firmware is going to be unreliable 
because there can be several messages in flight from the firmware to GC, so you 
may be acting on old data if you just decide based on the last thing you read 
from the serial stream.

you have to keep track of the messages you have sent vs the messages that have 
been acknowledged as being completed. otherwise you don't know if the message 
you are reacting to is the most recent one sent, or one from a few exchanges 
back.

BarbourSmith, 2017-06-06:
Great deduction. This sounds right to me. I didn't like the GRBL approach of counting the number of characters sent because an assumption is made about the length of buffer in the firmware and because you can get out of sync if the gcode sender program misses an "OK". My thinking was that if we're always printing the actual number of characters available things  will stay in sinc. That may have been over thinking things, so let's re-evaluate.

@blurfl is @davidelang correct that you see **'the unable to check buffer'** message when data is lost? Is it every time?

davidelang, 2017-06-06:
On Tue, 6 Jun 2017, BarbourSmith wrote:

> @blurfl is @davidelang correct that you see **'the unable to check buffer'** message when data is lost? Is it every time?

it happened at least once in the examples posted here.

blurfl, 2017-06-06:
I haven't been looking for the 'unable...' message, I will run some tests and report. To be sure I'm looking in the correct place - I'll need to run GC from the terminal  command line, and that message will be seen in the terminal command line window, yes?

blurfl, 2017-06-06:
Ok, I just loaded the current releases and started a run. The issue arose with no sight of 'Unable...' in log.txt or the terminal command line window.

blurfl, 2017-06-06:
I've snipped the pertinent sections from the serial echo, the terminal window and log.txt with the error line indicated. Note that in log.txt, in the PosError lines the third value is ringbuffer free space.

Serial echo:
G1 X16.0 Y106.494923858
G1 X15.5659898477 Y106.456852792
G1 X15.2347715736 Y108.339593909 F1500
G1 X14.4923857868 Y108.139593909 F1500
G1 X13.7969543147 Y107.817258883
G1 X13.1675126904 Y107.375634518
G1 X12.6243654822 Y106.83248731
G1 X12.1827411168 Y106.20304G1 X11.6598984772 Y104.763959391   <--------------
G1 X11.5939086294 Y104.0
G1 X11.6598984772 Y103.236040609
G1 X11.8604060914 Y102.492385787
G1 X12.1827411168 Y101.796954315

Terminal window:
   ding: G1 X16.0 Y106.494923858
   ding: G1 X15.5659898477 Y106.456852792
   ding: G1 X15.2347715736 Y108.339593909 F1500
   ding: G1 X14.4923857868 Y108.139593909 F1500
   ding: G1 X13.7969543147 Y107.817258883
   ding: G1 X13.1675126904 Y107.375634518
   ding: G1 X12.6243654822 Y106.83248731
   ding: G1 X12.1827411168 Y106.203045685     <----------------
   ding: G1 X11.8604060914 Y105.507614213
   ding: G1 X11.6598984772 Y104.763959391
   ding: G1 X11.5939086294 Y104.0
   ding: G1 X11.6598984772 Y103.236040609
   ding: G1 X11.8604060914 Y102.492385787
   ding: G1 X12.1827411168 Y101.796954315

Log.txt capture:
[PosError:0.51,0.30,176]
G1 X16.0 Y106.494923858
[PosError:0.43,0.38,170]
G1 X15.5659898477 Y106.456852792
<Idle,MPos:16.00,106.49,-1.82,WPos:0.000,0.000,0.000>
[PosError:0.19,0.27,170]
[PosError:0.29,0.53,176]
G1 X15.2347715736 Y108.339593909 F1500
[PosError:0.49,0.07,219]
G1 X14.4923857868 Y108.139593909 F1500
<Idle,MPos:15.23,108.34,-1.82,WPos:0.000,0.000,0.000>
[PosError:0.19,0.02,219]
[PosError:0.44,0.57,256]
G1 X13.7969543147 Y107.817258883
<Idle,MPos:13.80,107.82,-1.82,WPos:0.000,0.000,0.000>
[PosError:0.06,0.11,256]
[PosError:0.01,0.06,255]
G1 X13.1675126904 Y107.375634518
[PosError:0.17,0.74,228]
G1 X12.6243654822 Y106.83248731
[PosError:0.17,0.85,256]
G1 X12.1827411168 Y106.20304                              <----------------
<Idle,MPos:12.62,106.83,-1.82,WPos:0.000,0.000,0.000>
[PosError:0.03,0.57,256]
G1 X11.6598984772 Y104.763959391
[PosError:0.12,1.12,219]
G1 X11.5939086294 Y104.0
<Idle,MPos:11.65,104.62,-1.82,WPos:0.000,0.000,0.000>
[PosError:0.03,0.83,182]
[PosError:0.30,0.79,219]
G1 X11.6598984772 Y103.236040609
[PosError:0.32,0.54,183]
G1 X11.8604060914 Y102.492385787
[PosError:0.37,0.40,183]
G1 X12.1827411168 Y101.796954315

BarbourSmith, 2017-06-06:
Woah, that's interesting. So the log shows the correct line, but the serial echo shows the line being corrupted. What is the serial echo? Is that a capture of the serial communication?

I'm going to make a branch which adds a final term to each line indicating the correct number of characters which will be checked on the firmware side. I think that could help us track down what's happening and what causes it.

blurfl, 2017-06-06:
I added a line after Serial.read(c) to print c out Serial3 which is captured by a different USBserial connection. Looking at that serial stream, there are no characters between the digits and the G1 in the middle of the line. The two G1 commands in that line are broken out programmatically by the firmware.

BarbourSmith, 2017-06-06:
Wow! Great debugging tool!

So we know pretty conclusively at this point that the issue is in Ground Control because we've seen that
1) As soon as the Arduino is reading characters, there are characters missing.
2) Using universal gcode sender fixes the issue.

Right?

blurfl, 2017-06-06:
Exactly right👌

[74 more replies]

closed 2018-01-10 by BarbourSmith