MaslowCNC/GroundControl #301: Kivy installation fails to build on OSX 10.12.1: Clang error

braedenf, 2017-05-28:
I am having an issue compiling GroundControl on my Macbook. The issue is with Kivy failing to compile due to Clang errors: _error: command 'clang' failed with exit status 1_.

Is anyone else experiencing this issue?

BarbourSmith, 2017-05-28:
Yes! I spent like a week trying to figure that out and gave up. If you find a solution, I would love to know.

braedenf, 2017-05-28:
I solved the problem: All I had to do was install the latest Command Line Tools 8.3.2 for XCode, then it installed with the steps in the readme no problems whatsoever!

BarbourSmith, 2017-05-29:
Darn, I tried that with no luck. At least it worked for you 👍 

Ok to close the issue?

braedenf, 2017-05-29:
Yeah, I repeated my steps on two other macs but got another issue with pygame not windowing properly, however Kivy compiled fine. Have you tried deleting the command-line tools filder before re-installing it?

andymillernz, 2017-06-01:
On the issue with pygame and the window resizing issue on OSX I've added a note to the bottom of the wiki [mac install instructions](https://github.com/MaslowCNC/GroundControl/wiki/Mac) on how to ensure you are using SDL2 and not pygame for the Window manager - see [GL Reloading Section](https://kivy.org/docs/api-kivy.graphics.html) in the Kivy docs..

BarbourSmith, 2017-06-02:
Great! Thanks for updating the docs!

braedenf, 2017-06-02:
Yeah, I fixed the issue as well by doing: pip install https://github.com/kivy/kivy/archive/master.zip then it could find SDL2 fine :)

mrfugu, 2017-08-03:
This issue can be closed.

closed 2017-08-03 by BarbourSmith