MaslowCNC/GroundControl #309: Ringbuffer Possible Buffer Overflow Bug

andymillernz, 2017-06-02:
Have been looking over the Firmware source and seems to me there is an overflow bug in Ringbuffer. 

This is untested and may not be causing any issues however thought I'd point it out. And I could certainly be reading this wrong.......

 In _Ringbuffer.h_ `_buffer` is allocated to 256 bytes [0..255] however `_incrementEnd` (and `_incrementBeginning`) both allow the index to be set up to 256.  This allows `Ringbuffer::write` (and `::read`) to access `_buffer[256]` which is one position beyond the end of the allocated buffer.

I'd suggest changing `RingBuffer.h` to read `char _buffer[BUFFERSIZE+1];` as this makes reading the rest of the code 'easier' (instead of having `BUFFERSIZE-1` everywhere)..

davidelang, 2017-06-02:
Great timing, we were starting to investigate a data loss problem a few hours 
ago, this probably accounts for it.
k

andymillernz, 2017-06-02:
A case in Ringbuffer.cpp where the buffer overflow isn't detected

    void RingBuffer::_incrementEnd(){ 
        if  (_endOfString + 1 == _beginningOfString) || 
            (_endOfString==BUFFERSIZE && _beginningOfString==0) {

blurfl, 2017-06-02:
Making the recommended change to Ringbuffer.h did not fix the issue in #300. This is still a valid thing to look at, though.

blurfl, 2017-06-02:
Reading through _incrementBeginning and  _incrementEnd, it seems to me that they both test correctly for less than BUFFERSIZE: 
if (_beginningOfString < BUFFERSIZE)
and
if (_endOfString < BUFFERSIZE)
Aren't they correctly using BUFFERSIZE-1 as the upper limit here? If so, is the BUFFERSIZE+1 patch needed?

andymillernz, 2017-06-02:
Assume `_beginningOfString` is 255 coming into the `_incrementBeginning` function.  

It passes the `<BUFFSIZE` check and so it incremented to be 256..  

A call is then made to 'read',  which will reference ` _buffer[256]` which is an undefined memory area.

Much worse on the _endOfString/write case as writing into undefined space

blurfl, 2017-06-02:
I see, you're right. We should look closely at other places BUFFERSIZE is used as well.

BarbourSmith, 2017-06-02:
Firmware pull request [#236](https://github.com/MaslowCNC/Firmware/pull/236) put in the -1 to constrain us to the right range (great catch).

Are we still missing anything?

andymillernz, 2017-06-02:
I've posted a note against the [Pull request](https://github.com/MaslowCNC/Firmware/pull/236)

closed 2018-01-10 by BarbourSmith