MaslowCNC/GroundControl #310: Truncate gcode values at 4 digits beyond the decimal point.

blurfl, 2017-06-05:
Some gcode creation programs write values to 8 or 9 digits beyond the decimal point.

 This is way beyond our precision and impacts serial transmission, buffering and string handling.
GroundControl should truncate those values and keeping four digits beyond the decimal point seems about right to me.  

Universal Gcode Sender has a configuration option to choose the number of digits but that could be an enhancement for the future.

davidelang, 2017-06-05:
another useful thing would be to calculate the distance between one location and 
the next, and if the difference is too small, skip the command (will really help 
when the source used too many line segments for a curve)

k

johnbenweber, 2017-07-08:
Wrote a fix for truncating floats to 4 digits, see PR #339.

blurfl, 2017-07-12:
Nicely done, thanks for all the effort!

closed 2017-07-12 by blurfl