MaslowCNC/GroundControl #324: Calibration Routine glitch

blsteinhauer88, 2017-06-16:
With the latest fw/GC my sled is having trouble finding center, I have an error ring and it just climbs the board. I decided to Wipe the eprom, reload the FW and calibrate like new. When I get the the first step to align the gear, the left motor will move 1degree, but the right motor turns about a full turn of the gear. I cannot align the gear to finish the steps.  I closed, reloaded all again, and it is confirmed that the right motor 1 degree cw or ccw movement it turning a full turn of the motor.

blsteinhauer88, 2017-06-16:
Additional, It does not matter if you press cw or ccw it turns cw a full turn.

blurfl, 2017-06-16:
I too see the error ring get larger the farther from the center. GC76, Frimware76.

blsteinhauer88, 2017-06-16:
Try using the calibration and see if the right motor turn full turn instead of 1 degree to confirm.

blurfl, 2017-06-16:
I find that the right motor doesn't move in response to either CW or CCW button; the lift motor responds as expected. Too, the error circle grows as the sled leaves the center, shrinks as it returns to center.

blsteinhauer88, 2017-06-16:
That is what I have had this morning too. Thanks

BarbourSmith, 2017-06-16:
I had this behavior last week, but I thought I fixed it. Thank you for letting me know it's still a problem. I'll dig into it again and see what I can find.

BarbourSmith, 2017-06-16:
As a quick work around something I've seen fix this issue is to run the manual calibration once, then do the automatic calibration

blsteinhauer88, 2017-06-16:
I tried manual calibration.  Still have the problem with the motor turning full turn and only in the cw motion.

GeroBH, 2017-06-16:
I did not try the GC76, Frimware76 yet, was hoping to do testpattern tomorrow. With the two previous GC/FW I had similar issues with aligning the sprockets. Chatter on the right motor while turning the left, that would result eventually in bringing the sprocket off the 12:00. Trying to align right sprocket moved the left more clear and with less chatter.

BarbourSmith, 2017-06-16:
@blsteinhauer88 You are running the v0.76 firmware and Ground Control, right?

I was seeing this same exact behavior sometimes before Wednesday and it was one of the things I tried to fix in the Wednesday release (although it seems like I might not have got it completely sorted)

BarbourSmith, 2017-06-16:
Do you have any advice for how to make the problem happen? I've been trying to make the problem happen again this morning and I can't get it to behave that way. I've tried unplugging the machine while it's moving, pressing 'stop' part way through the calibration process and starting over.

Mine doesn't behave perfectly, sometimes one of the motors will twitch a little, but I'm not able to get it to do the full rotation behavior

blsteinhauer88, 2017-06-16:
I loaded GC 75 with no change.  I loaded Firmware 73, and GC 76 works.  The problem in is the new firmware.

BarbourSmith, 2017-06-16:
Oh woah, thanks! That's great to know. So I actually broke it this week, not made it better? Ooops! 😨

BarbourSmith, 2017-06-16:
I'm at a little bit of a loss as to what to do next, because until I can replicate the issue it's going to be really hard for me to figure out what's going on.

@GeroBH will you try doing the calibration process with v0.76 also and see if you can give us more insight into what is happening?

BarbourSmith, 2017-06-16:
@blurfl Do you have any advice? It seems like everyone might be seeing the issue EXCEPT me

blurfl, 2017-06-16:
I’m away from the shop until later in the day. I’ll see what I can find then.

blsteinhauer88, 2017-06-16:
@bar, did you get it to happen?  Today I had to restart the computer for an update to windows.  I loaded the new .76 versions, and It told me it did not know where it was, and to run the calibrate.  I ran the calibrate machine and immediately had the problem zeroing the gears.  I was able to compete the calibration with GC 76 and Firmware 73.  I will reload  FW 76 and see how it performs.

BarbourSmith, 2017-06-16:
After I couldn't get the issue to happen I got sucked in to #300 , I'll have another go right now

bar, 2017-06-16:
@blsteinhauer88 mhh?

blsteinhauer88, 2017-06-16:
[image]
Bar, I ran David Lang's test pattern, (results to post later) 
I uploaded firmware, .76.  I moved it only 10 inches.  I took a screenshot.  You can see it is off by .2 both directions.  I don't know if it will zero itself.

bar, 2017-06-16:
@blsteinhauer88 you are talking to the wrong person...

GeroBH, 2017-06-16:
Will have some hours time to run calibration and hopefully some cuts with a brand new Linux Manjaro in about 8 hours.

blsteinhauer88, 2017-06-16:
@bar, sorry wrong dude.

blsteinhauer88, 2017-06-17:
[image]
Here is what happend when I pushed Home, and my sled went to center, then traversed to about +5,+4.

BarbourSmith, 2017-06-17:
Hmm. I replaced the error circle with a second set of cross hairs so now Ground Control will give us the white cross hairs to show where the machine wants to be, and the red cross hairs to give us an estimate of where it actually is. If you download the very latest firmware and GC from GitHub it might give us a better sense of what is going on
[image]

blurfl, 2017-06-17:
I noticed that on the files I'm testing - I like it!

BarbourSmith, 2017-06-17:
Did it give any better information about what's going on? When you get that big error circle, where is the red cross hairs?

blsteinhauer88, 2017-06-17:
Cool. loaded GCmaster and this is what happened. 
[image]

blurfl, 2017-06-17:
I had that, did the manual chain length cal and it straightened out. It also might be a fluke of the screen draw, if white is drawn first, red covers it over. They lined up differently depending on my zoom level.
I found that the red circle and crosshairs gave interesting info, maybe alarming. If the red accurately indicates where the cutter is and the white is where it should be at that same moment, I'm seeing differences. Kind of hard to decide how big as there's no scale to go by. I found myself wishing that there was a 'keep the crosshairs centered' mode so I could zoom in and watch the cut up close -future enhancement-. Also wished for a little bar graph of the error value, with a high-water marker -future enhancement-.

BarbourSmith, 2017-06-17:
Both great suggestions!

The inner circle on the cross hairs is 1mm so when you are zoomed way in you can still get a sense of scale.

I agree that the error is more than I expected. It looks like because the red cross hairs are always lagging behind the white ones, it almost follows the correct path, but we're probably cutting some corners in ways we don't want to be. Certainly something to work on!

blsteinhauer88, 2017-06-17:
To continue where I left off @BarbourSmith, my last screenshot was just installing the GC master.  It showed the red cross hairs in about the postition the sled was sitting.  I hit the home button and GC did this:
[image]

BUT! the sled did this and I had to pull power so it would not tear the frame and sled apart.
[image]

I am going to back load FW .73

GeroBH, 2017-06-17:
Calibration was no fun with the new linux and FW0.76.GC0.76. No chance to complete. 2 times I had negative values for Motor Spacing and vertical offset and very diferent ones. https://dl.dropboxusercontent.com/s/g4rb024bin6t40c/calib1.jpg?dl=0 https://dl.dropboxusercontent.com/s/f5s5o2qwaamrr9g/calib2.jpg?dl=0
Retracting left chain would somtimes extend very short and then retract. You need to look close to see that. After manual chain length and entering the dimensions buy hand, I can move the sled around. No trace of the red circle here. Z-Moves make both motors chatter to an extent that the chains are chivering.

BarbourSmith, 2017-06-17:
I've got some great news. Using Hannah's computer and the other machine, I'm seeing the same (I think) behavior you guys are. Hopefully I can track down what's going on this way.

BarbourSmith, 2017-06-17:
I think I see what's going on. I'm working on a fix right now 👍

BarbourSmith, 2017-06-17:
I've got something that took me through the calibration process twice on the system that didn't work before. 

When anyone has a chance, check out the latest firmware and Ground Control from Git Hub and let me know if it fixes the issue for you too. 

The issue was with the way the calibration process handled unexpected starting conditions. Last week I added some code to the firmware which would try to catch when something funky was going on and reset the values, but it was running at times we didn't want it to and messing things up.

Now the process is handled by Ground Control. When you click the "Begin" button, the machine is set to a known state, so in theory we should always start the calibration process with everything set right.

Let me know if it works!

GeroBH, 2017-06-18:
@Bar "latest firmware and Ground Control from Git Hub" <-- still don't know if that is the the green download button, or the latest under 'releases' (assume they are the same) and if the .tar.gz always gets updated the same time as the .zip, kindly advise. Got used to go for the 'releases' as they have the version in the compressed file name and I am archiving now.

BarbourSmith, 2017-06-18:
Good question 😀 

The "latest" is always the green download button, the latest version under "releases" will be the version from the last Wednesday

The .tar.gz and .zip files will always be updated at the same time (they are generated automatically) when the release comes out.

davidelang, 2017-06-18:
the releases are not the latest, they are what is tagged as a release on 
wednesdays

The green 'clone or download' will grab the very latest version that has been 
pushed to github (bar may have a newer copy on his machine, but github has no 
way of knowing that)

On Sat, 17 Jun 2017, Gero wrote:

> @Bar "latest firmware and Ground Control from Git Hub" <-- still don't know if that is the the green download button, or the latest under 'releases' (assume they are the same) and if the .tar.gz [...]

bar, 2017-06-18:
@GeroBH I have no idea what you are talking about :) you are talking to the wrong person

davidelang, 2017-06-18:
apologies, please be patient with us (and keep reminding us as someone new makes the mistake of doing @bar instead of @BarbourSmith  (especially those of us who respond via e-mail and so don't have the github web interface prompting is for auto-completion of the names as we type)

blsteinhauer88, 2017-06-19:
Bar, I loaded your latest FW and GC.  The calibration routine seemed to work.  The right motor did not do full turns.  I did not finish the routine and went back to GC.  I told the sled to travel up 4 inches and each time it went down.  
[image]
It ended up here.  When I hit return to center, it also travels down.  ????

BarbourSmith, 2017-06-19:
What was happening before was that not only were the motors not turning the right amount, the values that were being stored in settings were way off (I had a negative spacing between the motors on mine)

I would try either manually entering values you know are good in the settings, or going through the whole calibration process to make sure that the machine has good values

BarbourSmith, 2017-06-21:
Would anyone be willing to take a look at the version currently on GitHub to check if the calibration process is working?

I've tested it out on my end, but having a second set of eyes on it would be really valuable.

You'll know you are running the right version because the version number will be 0.77

[image]

Thanks everyone for your patience while I've been sorting this one out!

blurfl, 2017-06-21:
I’ll give it a go, report back in a bit.

blurfl, 2017-06-21:
Part way through, I notice that in step 7 when raising and lowering z, the sled also moves in the xy direction, small seemingly random jumps. Z axis moves as desired, though. FWIW, I noticed this same behavior before I started the cal steps, just setting z when I first started GC77/FW77.
So fr the other steps are as normal, now for the cuts!

blurfl, 2017-06-21:
I just completed the cal sequence, three cuts to arrive at 0.3mm difference between horizontal and vertical.

ladyhawke13, 2017-06-23:
During calibration my left motor will only adjust CCW. When I select move left move CW 1 degree the motors hums but no actual movement.

BarbourSmith, 2017-06-28:
@ladyhawke13 are you still seeing this issue?

ladyhawke13, 2017-06-28:
It was consistently doing that through my calibration - unfortunately I can't test it to see if it's still doing it at the moment - everything's in pieces! However, I'm hoping to get everything back together by the weekend so hopefully I can let you know in a few days.

BarbourSmith, 2017-06-28:
Perfect! Keep me posted if the issue persists, and we'll track it down

closed 2018-01-10 by BarbourSmith