MaslowCNC/GroundControl #337: Add version as top level variable in main.py for Android APK building

jbnimble, 2017-07-04:
When building an Android APK out of GroundControl it is required to have a "version" variable at the top level in the main.py. See the example in the [building executables](https://github.com/MaslowCNC/GroundControl/wiki/Building-Executables) for Android documents.

Example:
```python
version = 0.78
```
or even a dynamic version would probably be better:
```python
version = str(self.data.version)
```

jbnimble, 2017-07-04:
Turns out the latest version of Kivy and Buildozer no longer use the main.py "version" variable and instead use the buildozer.spec "version" property.

closed 2017-07-04 by jbnimble

BarbourSmith, 2017-07-05:
👍 👍 👍 

Let me know if you get it to build, it's on my todo list but a ways off 😀

jbnimble, 2017-07-05:
@BarbourSmith, Built fine, updated wiki instructions. Able too execute on older Android tablet and opened various menus. Did not try to load gcode, and no hardware to test serial connection yet. Thinking of writing a script that takes care of editing the buildozer.spec and running the buildozer commands. Where could a script pull the version number?