MaslowCNC/GroundControl #356: Calibration Machine Dimensions Step 8

jbnimble, 2017-07-20:
After running calibration several times, the place where the bit cuts becomes overlapping with previous calibration runs. Allow an offset, the ability to scale the marks larger or smaller, or some other method so that the bit does not cut over top of previous calibration cuts.

blurfl, 2017-07-23:
I’ve run into this too. Flipping l-r or u-d might do it as well.

BarbourSmith, 2017-07-25:
This is an issue, but I'm not sure what the solution is. Running the calibration centered helps a lot with keeping everything consistent. I'm worried if we add an offset option that how much offset is added will affect the final calibration.

blurfl, 2017-07-25:
How about letting the user choose between the current step down pattern and step up? Or limit the offset to 10mm?

BarbourSmith, 2017-07-25:
A limited offset is a good idea

BarbourSmith, 2018-01-10:
There is now an offset slider option 👍

closed 2018-01-10 by BarbourSmith