MaslowCNC/GroundControl #365: Work Area Zoom

jbnimble, 2017-07-20:
Using Linux (Ubuntu latest and a laptop with a touchpad) I cannot zoom in and zoom out using the page up and page down keyboard keys.

davidelang, 2017-07-20:
on Linux, I think you need to use the mouse scroll wheel.
k

jbnimble, 2017-07-20:
Is there another option, my laptop is a touchpad which uses multi-touch and does not work correctly in Ground Control, and my external mouse is a Logitech where the user moves the mouse by rolling a ball mounted on the top, and so no scroll wheel. Can some other key be mapped?

davidelang, 2017-07-20:
on my linux laptop a two finger up/down scroll zooms

this would be part of the kivy config

BarbourSmith, 2017-07-20:
It might be worth trying to map a different key to be zoom. Under settings you can change which keys make the zoom happen, it's possible that page up and down are different on linux for some reason.

[image]

jbnimble, 2017-08-21:
Tried to find what the magic string is to bind keys in linux, but no joy.
I guess this is the correct documentation?
https://kivy.org/docs/api-kivy.modules.html#module-kivy.modules
https://kivy.org/docs/api-kivy.modules.keybinding.html?highlight=binding#

BarbourSmith, 2018-01-10:
I am going to close this although it isn't really resolved it hasn't been touched in months and I'm not sure it's a GC issue 😐

closed 2018-01-10 by BarbourSmith