MaslowCNC/GroundControl #366: Hold button stops machine immediately

jbnimble, 2017-07-20:
The "Hold" button only takes effect after the next G-Code line is ready to execute, it would be nice if the hold button could stop the currently executing line of G-Code and resume on "Continue". Otherwise long straight lines, and slow Z-Axis moves take a long time before the machine actually stops.

I don't understand the firmware implications, but I assume this might mean taking the given G-Code instruction A to B and converting it into an A1 to B, where A1 is an on the fly generated G-Code instruction that moves from where the hold took effect during the middle of the A move and then completes the rest of the A instruction to begin the B instruction.

davidelang, 2017-07-20:
what this would require is to have two stages of processing input

First, something would have to understand the input well enough to find the end 
of a message (and understand comments so it doesn't trigger on comment text) and 
watch for pause/halt commands

The second level would need to parse the gcode into the needed movement

(insert standard rant here about reinventing the wheel instead of leveraging 
existing CNC firmware that already solves these and other problems :) )

davidelang, 2017-07-20:
also, once we involve knowledge of acceleration, hitting 'stop' will mean that 
you are halting immediately, not giving the machine a chance to slow down, so 
you cannot seamlessly restart the cut
k

JDogHerman, 2017-07-27:
Since it is a closed loop system we should know the percentage of the line run and it should be possible to continue without much issue. The router's motions are relatively slow and acceleration would have some impact but could be acceptable.

BarbourSmith, 2018-01-10:
I am going to close this, while I think it's a good idea it's going to be tough to implement and I feel like it might make it difficult to resume a paused program later

closed 2018-01-10 by BarbourSmith