MaslowCNC/GroundControl #373: Return to home causes the motors to continue after being held

JDogHerman, 2017-07-24:
v0.81

Ground Control was running some g-code. I put it in hold after it asked for a depth change. I then told Ground control to return to home. It returned to home then continued to run gcode from the new position. 👎

BarbourSmith, 2017-08-22:
I believe that this issue has been fixed by the removal of the "Hold" option in the z-axis popup. Would you agree?

JDogHerman, 2017-08-25:
Without a hold button it seems to be fixed but, how would I cancel the job?

krkeegan, 2017-09-01:
Clicking Stop will cancel the job.

I think this isssue is likely ripe for closing

JDogHerman, 2017-09-01:
If the z axis dialog is up there is no way that I see to cancel the job. The stop button along with all other buttons (except for continue) are inaccessible. The only way I see to actually quit would be to close ground control.

krkeegan, 2017-09-01:
Ahh, yes, I see the issue now.  I guess we need a stop button in the zaxis dialog?

BarbourSmith, 2018-01-10:
This has been added 👍 👍 🌹

closed 2018-01-10 by BarbourSmith