MaslowCNC/GroundControl #375: Shortcut or button to center work area

jbnimble, 2017-07-25:
Using a laptop with a touchpad sometimes causes the work area to move "off screen", and then I can't find it again. I would like a way to to cause the work area to be in the middle of the viewing area.

davidelang, 2017-07-25:
ideally, such a 'find me' button would also zoom so that the entire part that's 
being cut fills the screen (not the entire work area as the piece being cut 
could be very small)

blurfl, 2017-07-25:
‘Actions/Reset View’ accomplishes part of this. There is no keyboard equivalent. Adding more buttons to the front screen gets in the way for users with smaller displays, though.

jbnimble, 2017-07-25:
Does actions > reset view effect the currently running cut?

blurfl, 2017-07-25:
I don’t know for sure, but I don’t think so...

BarbourSmith, 2017-07-25:
No, it shouldn't have an effect on the cut, it just moves the preview to the center

jbnimble, 2017-07-25:
If that is the case, then that is good enough for me, and this issue can be closed, unless someone else needs more.

closed 2017-07-25 by BarbourSmith