MaslowCNC/GroundControl #376: Permanent frame parts cut incorrectly

mike929t, 2017-07-25:
The permanent frame is still the thing that I most to build with Maslow! I took another run at it two weeks ago, and failed again. On July 11th I used the temporary Beta frame that I built months ago, along with the then-latest versions of the firmware and GC. I grabbed the All Parts .SVG file and ran it through MakerCAM to create the Gcode. When I ran GC to cut all the parts, several errors occurred along the way, and then the router suddenly headed "straight north", cutting a path toward the top of the plywood until I manually stopped it. See the attached images for details (to which I've added text explanations). I'm almost positive that at least some of the problems are the result of characters getting dropped from the data stream. May I suggest that you distribute Gcode files for each of the individual parts for the permanent frame? I think that if I can cut them one at a time, and skip the MakerCAM file format conversion step, I'll stand a much greater chance of success. Thanks, Mike Thompson
[image]
[image]
[image]
[image]

mrfugu, 2017-07-25:
The [MaslowCNC Forums](http://forums.maslowcnc.com/c/software-issues) is probably the best place to address this issue. I believe there are multiple threads there discussing this known issue. 

(please close this issue here.)

mike929t, 2017-07-25:
If there are "multiple threads there (MaslowCNC Forums) discussing this known issue", then will you please direct me to them? I can't find them.

I think there are a couple problems here. The first is that characters in the Gcode are getting lost - likely evidence of a buffer overrun problem that Bar has been chasing for a long time. The second problem is with the method of distributing the plans/code for cutting the parts for the permanent frame. Jamming them all into one .SVG file just isn't good enough. It leaves the user in a position of having to cut all the parts at once (which, as I've shown, Maslow doesn't do correctly), or else figure out how to dis-assemble the .SVG in order to be able to cut them individually. I've had good luck with creating simple parts in Inkscape, converting them to Gcode with MakerCAM, and then getting Maslow to cut them. I'm having no luck trying to cut all the parts at one time with a single Gcode file.

BarbourSmith, 2017-07-25:
I think this is an appropriate place for the software issue side of things. I know we were seeing these types of buffering issues with earlier versions of the firmware/software, but I thought we had fixed them. 

Which firmware/ground control versions did this happen with?

mrfugu, 2017-07-26:
apologies all around. i was trying to direct troubleshooting towards the Forum, and this 'issue' area for software bug documentation. 

(This appeared to me to be similar to the Z-axis/lack of memory issues reported by others. )

mike929t, 2017-07-27:
*Mr Fugu,*

No worries - you were just trying to be helpful. Thanks for pointing me at
the latest .ZIP file containing the .SVG files for the permanent frame. The
old .ZIP file version that was been using is missing couple of the
individual .SVG files.

By the way, nice puffer fish picture!

*Bar,*

I'm using versions of the firmware and GC that I downloaded on June 7th. It
was shortly after you said that you thought you had squashed the bug that
was resulting in data being dropped when big Gcode files were run on fast
machines.

The firmware version is V0.73. How do I tell what version of GC it is?
(Please consider this a suggestion that an About button that shows the GC
version number, and/or the display of the version number at startup, would
be nice additions to GC.)

Thanks,
Mike Thompson

mike929t, 2017-07-27:
...and here I'm attaching the image files of the failure. The first one
shows the GC display, with an arrow pointing to a G1 command that looks to
have been damaged by data loss. The next three are pictures of the plywood.
Note: (1) the vertical path where the router "headed north" when trying to
execute the bad G1 command, and (2) several places where the router went
somewhat astray on one or more of the 5 passes (1/8" deeper on the Z axis
each pass) that it made.

Also note that the digits ("1", "2", "3"), that were to be carved into the
three different types of angle bracket, were cut to full depth (all 5
passes of the router), though I'm pretty sure they were only supposed to be
cut "skin deep". This can probably be considered user error (I should have
thought to tell MakerCAM that the depth for the digits should be different
than the depth for the outline of the part), or else considered to be an
assembly documentation error.

Thanks,
MikeT

BarbourSmith, 2017-07-27:
I'm fairly certain (but let me know if I'm wrong) that these problems have been fixed in the latest version of the firmware/Ground Control. Even better, your excellent suggestions about making the version numbers more prominent have been added too. 

It's nice to be reminded that things do improve 😁 

Because things are still changing so quickly I'd recommend updating the software/firmware as a good first step because the odds are that someone else has run into the same issue and possibly fixed it

[image]

[image]

mike929t, 2017-07-27:
OK, thanks Bar. I'll go get another sheet of plywood, download the latest
versions of the firmware & GC, and give it another go.

Take care,
MikeT

BarbourSmith, 2017-07-27:
It might be worth testing on some corner of that one just to be sure that the issue is resolved, rather than risk a whole new sheet

closed 2017-08-22 by BarbourSmith