MaslowCNC/GroundControl #377: GC Settings ‘Valid File Extensions’ list change not honored

blurfl, 2017-07-25:
Adding an item (.cnc) to the list does not show those files in the load dialog. Restarting GC makes no difference.

BarbourSmith, 2017-07-25:
I think that #265 is very related

blurfl, 2017-07-25:
The FileChooser filter list is currently hard-coded in groundcontrol.kv, see #265. @johnbenweber suggests a solution. Is data.py the appropriate place to declare a global to declare the global array he suggests, or would it be better to create it somewhere else? I think that once created, the FileChooser could use it for the filter, yes?

BarbourSmith, 2017-07-25:
I'd reference the settings directly when the list is needed using data.config. It's really slow to read from the settings, but the operation would only happen once each time new files are going to be opened.

Something else that might be worth bringing up here is that I've been thinking about abandoning the kivy file chooser in favor of the TK one. The TK one opens much more quickly and looks the same as every other "open file" window in the OS. https://pythonspot.com/en/tk-file-dialogs/

The main drawback is that TK doesn't work on Android which I'd like to support some day so the structure would have to be something like:

If operating system is windows linux or mac:
        Import TK
        Use the TK file chooser
else:
        Use the Kivy file chooser

johnbenweber, 2017-07-25:
@BarbourSmith I agree, I find the kivy file chooser kind of annoying. You could put the `import tk` call in a `try...catch` block with the kivy filechooser being used if that fails.

blurfl, 2017-07-25:
Very much agree that the kivy file chooser is annoying. This goes pretty far beyond my comfort level, though - I'll stand back, watch and learn :).

johnbenweber, 2017-07-25:
I'll add that this issue wouldn't be that difficult to fix. The reason the extensions setting wasn't used for the kivy filechooser filtering is that an array of file extensions is required but the setting is just one long string. The string would have to be broken up and converted to an array to be useful.

An easy temporary workaround is to just change the file extension on whatever file you are trying to load. A quick and dirty fix for the UI might be to disable the setting to indicate that changing it does nothing. It's more there at the moment to inform the user what file extensions they can expect to work.

blurfl, 2017-07-25:
You're right, there are bigger fiash to fry :). Implementing the TK file chooser could be an enhancement for later - shall I close this and open another as an enhancement?

BarbourSmith, 2017-07-25:
Love the try..except idea. Great recommendation @johnbenweber 

Can we just change the tags on this one to be "enhancement"? That way we won't loose the conversation which has all the tips for how to implement this.

blurfl, 2017-07-25:
Sure that will be best.

BarbourSmith, 2018-01-10:
It is valid now! 💯

closed 2018-01-10 by BarbourSmith