MaslowCNC/GroundControl #378: Job Border or corner function.

DanielDePauw, 2017-07-27:
It would be nice to have a button that would move the sled around the border of the job. This is feature would have 2 purposes:

1. You could test the outside edges of a cut before turning on the tool to ensure your job will not run off the edge of the material.
2. You can use it to position the material to best utilize the space and prevent waste.

I have seen this feature on laser cutter software and it becomes habit to test the borders of the job before starting to save material and prevent damage to the machine and humans.

This feature could be implemented with a button that cycles through each corner of the job when pressed. (Ex. pressing the button once moves the tool to the lower left corner, next press the tool moves to the lower right corner, then upper right, then upper left.

jbnimble, 2017-07-27:
Yes!

BarbourSmith, 2017-07-27:
Love it!

jbnimble, 2017-08-21:
This is possibly similar to #387?