MaslowCNC/GroundControl #382: Some on-screen type _very_ small

blurfl, 2017-07-31:
Not certain when this cropped up, but it was correct in release 0.82.
[image]

BarbourSmith, 2017-08-01:
Oh, I'm glad you pointed that out. I've been messing with exactly that today 😬

blurfl, 2017-08-14:
Furthermore, the top two error sliders draw part of their text ouside the box - only 'Error' shows for each.

davidelang, 2017-08-14:
what sort of system are you using? what screen resolution?

blurfl, 2017-08-14:
The system is OS X. The labels drawn with screen resolution 1024x640; anything large shows the issue.

blurfl, 2017-08-15:
> I think that the last time we saw something like this the solution was to
> switch from a font size to sp (scale-independent pixels)
>
>
> https://kivy.org/docs/api-kivy.metrics.html

I’ve played around and made some progress. I’ll post a commentwith details later, after I get four more post holes dug 🙂

blurfl, 2017-08-16:
I swear I had both Windows and OSX displaying the Simulation control labels correctly last night. Tonight Windows is displaying wrong. I'm unable to get these things pounded into github for a day or so, so some observations:

I did find that Window.size[0] is width, [1] is height. Printing window.size on my OSX system shows the retina display size:
window w, h
(5120, 2706)

The tiny text is fixed by font_size: '10sp' as you suggested

On OSX I was having luck formatting the simulation labels so:
            Label:
                text: "Motor Spacing\nError: 0mm"
                text_size: self.width, sp(self.height)
                font_size: (self.height - 6)/2
                halign: 'left'
                valign: 'middle'
                id: motorSpacingErrorLabel

Sorry for the disjointed message, I hope something in it is of use>

BarbourSmith, 2018-01-12:
Fixed in #565! Way to go @blurfl 👍 👍

closed 2018-01-12 by BarbourSmith