MaslowCNC/GroundControl #385: B03 data separated into two chunks causes Firmware/Kinematics.cpp to fail

blurfl, 2017-08-16:
See [this thread](http://forums.maslowcnc.com/t/impossible-to-cut-a-piece-without-losing-calibration/454?u=blurfl) on the forum. The B03 data was broken into two chunks to try to fit within the grbl 70 character limit. Kinematics.cpp tries to function with a partial set of configuration values and fails, putting the machine into an unconfigured state. Altering GC to send the B03 data in one go is a work-around.

krkeegan, 2017-08-16:
I though of the same issue.

The odd thing is that the first B03 message contains nearly everything required for the quadrilateral kinematics.  The only thing I see that it doesn't receive is the kinematics type, but as the default is quadrilateral, I don't see why this should matter.

krkeegan, 2017-08-16:
So I think the issue might be the change in kinematics.h

Previously we set realistic default values for all of the kinematics.  Now, most of the values are 1.  If anything calls kinematics.forward before the values are set, it will result in a failure.

Is this possible?

blurfl, 2017-08-16:
Yes, that is the nub of it. If we can't send all the values in one packet (which I've found does work), maybe kinematics shouldn't run until a flag is set by the updateSettings function to indicate that that all values have been received. Or instead of a single flag, set all values in kinematics.h to an impossible flag value (-999?) and make kinematics wait until each has been made rational.

BarbourSmith, 2017-08-16:
Right now B03 sends all the configuration to the machine and it's massive.

What if we split it into multiple commands so B03 sends the machine
dimensions, Bxx sends the PID values, Bxxx sends the encoder
resolution...etc?

davidelang, 2017-08-16:
that sounds like a good idea. set flags as each set arrives, and if you get a 
movement command before everything is there, generate a message saying what 
parts are missing.

blurfl, 2017-08-21:
This issue is solved by PR#386. 👍 💯

closed 2017-08-21 by blurfl