MaslowCNC/GroundControl #387: Create new function "mark" to facilitate locating stock and set Home functions

nwrancher, 2017-08-17:


BarbourSmith, 2017-08-17:
Thanks for filing the issue @nwrancher!

jbnimble, 2017-08-21:
Just realized #394 is probably a duplicate of this issue. Here is the text from that issue, and I'll close it.

> It would be very convenient to be able to mark the current sled position in gc, and be able to add multiple marks. This way if you want to avoid or line up to a location you can do so to best set the define home location for cutting the gcode at the expected location.

blurfl, 2017-08-22:
Would providing a ‘go to x,y absolute’ button/dialog satisfy the requirement? Currently the move buttons are relative to the present position. A button would prompt for x and y (assuming 0,0 is the Return to Center spot) and the sled would move to those coordinates. Some sort of input bounds checking would be good.

BarbourSmith, 2017-08-22:
Or what about an option to click somewhere on the drawing and send the sled there?

jbnimble, 2017-08-23:
The use case for me is to be able to move the sled to a position, then mark that in ground control. The point for me is to show in ground control any boundaries of previously cut portions that I don't want the next cut to overlap. I would want to be able to mark multiple positions.

blurfl, 2017-08-23:
So, to create ‘keep-out’ areas, yes? Would this be visual only, showing on the screen as shaded polygons?

jbnimble, 2017-08-23:
To mark in ground control reference points based on the current position of the sled crosshairs in ground control. They could be used for any purpose but my use case is to mark the edge of previously cut pieces. They could be crosshairs or any other shape, maybe even a horizontal or vertical line option, but when added the position is based on the current location of the sled crosshairs.

BarbourSmith, 2017-08-23:
Does #396 fulfill this need?

blurfl, 2017-08-23:
Nice work!  You’ve pulled a rabbit out of your 🎩 hat.

BarbourSmith, 2017-08-23:
Thanks! I think it was a really good suggestion. Now that we've got a menu which responds to location based clicks I bet there's all kinds of other things we could use it for like measuring the distance between two points 👍

closed 2017-09-15 by BarbourSmith