MaslowCNC/GroundControl #388: Slide out menu for additional buttons

jbnimble, 2017-08-18:
Is it possible to create a menu in kivy that slides over top of the workspace temporarily and offers additional navigational and other button options? There has been lots of give and take about which buttons should be on the screen because of real estate and priority concerns, but if there could be a set of buttons that are only viewable after opening a menu/slide over view, that would be very convenient.

Some of the buttons I'm thinking are the "reset view" and "zoom in/out", that would be terribly convenient to have accessible.

BarbourSmith, 2017-08-22:
The built in kivy modules don't give us great options for this but the garden has some good things. Here are two I think look like they might work for us:

https://www.youtube.com/watch?v=OGTOIzI7SFs

https://www.youtube.com/watch?v=zzMTpcUGG0Q

blurfl, 2017-08-22:
The drawers look promising. It would be good to hear from users running on a Pi touchscreen about this, and about the present GriundControl for that matter. That little screen is already pretty filled up with controls.

BarbourSmith, 2017-09-14:
I added the context sensitive menu to address this, do we want to add any more options to that menu? 

We could add a 'View' tree to it with 'zoom in', 'zoom out', and 'reset view' options?

Anything else?

closed 2018-01-10 by BarbourSmith