MaslowCNC/GroundControl #392: Change color of gcode visual line after cut

jbnimble, 2017-08-18:
The visualization of the gcode is always white, it would be neat if the color could change after the line has been executed.

blurfl, 2017-08-21:
How would lines that will receive multiple passes work?

jbnimble, 2017-08-21:
Good point, not sure. The idea was to give a quick visual way to tell how much is completed and how much is left to go.

blurfl, 2017-08-21:
There’s a % complete and line number number on the right. I think the percent is based on line number rather than the time it takes to cut any one line.

jbnimble, 2017-08-21:
It is certainly a visual eye candy feature, and not required for operation. If the consensus is there are better  things to spend time then I'm fine with closing it unresolved.

BarbourSmith, 2018-01-10:
I'm going to close this for now because I am worried that it would require a lot of ram when loading large files

closed 2018-01-10 by BarbourSmith