MaslowCNC/GroundControl #399: Clicking stop when already stopped causes the motor to turn on.

JDogHerman, 2017-08-25:
Observed with v 0.87 

The motors will sometimes just hum and other times the output shaft turns.

JDogHerman, 2017-08-25:
The position markers don't move during this time even-thou the motor output shaft turns.

krkeegan, 2017-09-01:
This may be related to the Firmware issue [#293](https://github.com/MaslowCNC/Firmware/issues/293).  Although I don't know why the motors would go from off, to back on again when pressing stop.

BarbourSmith, 2018-01-10:
Any command sent to the machine will cause the motors to energize for two seconds to hold their position...is this behavior wrong in the case of the stop command?

krkeegan, 2018-01-10:
We should check to see if this still happens.  I don't think clicking stop after the motors are stopped causes them to turn on anymore.

davidelang, 2018-01-10:
It seems like it would probably cause more grief to try and special case this.

BarbourSmith, 2018-01-10:
Best possible answer @krkeegan I just tested it and you are right pressing stop does not engage the motors anymore

closed 2018-01-10 by BarbourSmith