MaslowCNC/GroundControl #400: Escape button causes GroundControl to quit.

JDogHerman, 2017-08-25:
clicking the escape button...

```[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 643, in <module>
     GroundControlApp().run()
   File "C:\Python27\lib\site-packages\kivy\app.py", line 828, in run
     runTouchApp()
   File "C:\Python27\lib\site-packages\kivy\base.py", line 504, in runTouchApp
     EventLoop.window.mainloop()
   File "C:\Python27\lib\site-packages\kivy\core\window\window_sdl2.py", line 663, in mainloop
     self._mainloop()
   File "C:\Python27\lib\site-packages\kivy\core\window\window_sdl2.py", line 606, in _mainloop
     self.modifiers)
   File "kivy\_event.pyx", line 714, in kivy._event.EventDispatcher.dispatch (kivy\_event.c:8146)
   File "kivy\_event.pyx", line 1225, in kivy._event.EventObservers.dispatch (kivy\_event.c:14035)
   File "kivy\_event.pyx", line 1149, in kivy._event.EventObservers._dispatch (kivy\_event.c:13564)
   File "C:\Python27\lib\site-packages\kivy\app.py", line 1040, in _on_keyboard_settings
     return self.close_settings()
 TypeError: close_settings() takes exactly 2 arguments (1 given)

wrh3c, 2017-09-08:
I believe this has been fixed by pull request #408.  Now that the keyboard binding in GcodeCanvas is preserved even when popups are opened and closed, the Esc key is caught and ignored.

That said, we should consider setting the kivy exit_on_escape config token to 0 to prevent the Esc key from quitting GroundControl.  See the discussion at https://stackoverflow.com/questions/27590127/leaving-kivy-app-when-pressing-escape for more information.

BarbourSmith, 2017-09-08:
I think we might have wanted it to quit the program for those operating on raspberry pi where Ground Control launches full screen with no option to click an "X" in the corner.

Does anyone else remember that? @blurfl ? @TheRiflesSpiral ?

blurfl, 2017-09-08:
I don’t remember, and I’m not in a place to check. There is the quit button though.

BarbourSmith, 2017-09-08:
Great memory, I forgot that we also added the quit button. That should be enough.

TheRiflesSpiral, 2017-09-08:
Yep, this is how I remembered it. Esc to navigate back through menus, Quit button to quit the app.

wrh3c, 2017-09-08:
We broke the ability to use Esc to navigate back through menus with pull request #408.  I'll experiment to see how we can add that back.

From my experiments based on the issue report, it looks like Esc causes Ground Control to quit when trying to navigate back from the Settings menu.

TheRiflesSpiral, 2017-09-08:
Being primarily a Mac user I'm addicted to the Esc key. (it's one of the reasons I haven't upgraded to the new MBP) but I think the vast majority of users won't miss that.

wrh3c, 2017-09-09:
I submitted pull request #417 to enable the Esc key again in addition to fixing the original issue of quitting with the error "TypeError: close_settings() takes exactly 2 arguments (1 given)".

This also allows use of the F1 key to enter and exit the settings menu - a built-in behavior in kivy.  This was new to me!  There's a lot in kivy that I don't know, and I feel like I'm only scratching the surface.

BarbourSmith, 2017-09-15:
It looks to me like @wrh3c nailed it and I'm going to close the issue 😀

closed 2017-09-15 by BarbourSmith