MaslowCNC/GroundControl #401: Motors will continue to run even after GroundControl has closed.

JDogHerman, 2017-08-25:
If GC is closed while a job is being run the motors will continue to run. This seems like an important safety concern.

davidelang, 2017-08-25:
how would the firmware know that GC was closed? it's just following the last 
instruction it was sent, and doesn't know if GC was closed or if it's just 
about to send a new command (and it can take a while to finish executing a 
command)

Given that there is other software that can send commands to the maslow, it's 
not clear that the maslow should shutdown if GC exits anyway.

JDogHerman, 2017-08-25:
Maybe a heartbeat should be setup. This would monitor if communication was lost from the application closing or the USB becoming unplugged.

BarbourSmith, 2018-01-10:
If the USB becomes unplugged the machine will stop immediately, but if GC is closed it will finish executing the current line of gcode then shut itself down and save it's position. I feel like this is the desired behavior (it's the same as GRBL which is the industry standard right now). How do we feel about closing this one?

blurfl, 2018-01-10:
I agree, OK by me.

davidelang, 2018-01-10:
I think it makes a lot of sense to match grbl

BarbourSmith, 2018-01-10:
Oops, I didn't mean to reopen 😬

closed 2018-01-10 by BarbourSmith