MaslowCNC/GroundControl #407: Calibration Step #7 - Z-Axis

bdillahu, 2017-09-02:
I know somewhere in the instructions it mentions turning on the Z-Axis in settings if you have it... it would be REALLY nice if this was repeated on this step #7. Even better, just have a button here to click if you have a z-axis, or "skip" if you don't and let that turn on the settings.

I didn't turn on the setting, got to this step and 'assumed' it would work. What's bad is that since the motor controller is 'shared', the other motors buzz and make noise, making you think it's working. When I figured out it wasn't moving, I started taking things apart trying to debug it. Mechanical was fine, software was off.

BarbourSmith, 2017-09-05:
Love it, great suggestion!

BarbourSmith, 2017-09-14:
I think I've added the feature in #421 when you get a chance will you take a look and let me know if I'm missing anything @bdillahu ?

bdillahu, 2017-09-17:
Looks "nearly" right :-)

If you turn it off in settings and then turn it on during calibration, the settings screen doesn't pick up the change until you restart GroundControl. It appears the settings screen doesn't re-read the config file each time (?), but reads it at startup.

The value appeared correctly, so I think it's working except for that display glitch.

Bruce

BarbourSmith, 2017-09-17:
Very interesting! 

It looks like you are exactly right that the settings window doesn't reflect any changes made after Ground Control launches unless the change is made through the settings window.

It might be a tough thing to change because the settings window is built into Kivy, but I'll take a look and see what I can find.

BarbourSmith, 2017-09-25:
From doing a little digging it seems like fixing that display glitch would be a pretty big amount of work (probably involving committing to the KIvy project and waiting for them to come out with a new version). 

Would we feel OK about me closing this with the caveat that there is a known glitch with the way Kivy updates the settings panel, and that the next step would be to file a bug with Kivy if we really think it has to be fixed?

bdillahu, 2017-09-25:
I’m fine with it... really the ‘problem’ was getting through calibration and not having the z-axis enabled, which your fix did just fine. I was just being overly thorough and testing the other display screen :-) (it was actually the easiest way to see if your new switch actually did anything :-))

Bruce

closed 2017-09-25 by BarbourSmith