MaslowCNC/GroundControl #419: Mouse on Raspberry Pi only showing when clicked

guidobender, 2017-09-10:
Hi,
I just installed (first install, no comparison to older versions) the latest Kivy and Ground Control versions on a Raspberry Pi B to be used as the base machine for our Maslow CNC. I followed the instructions closely. The input devices on the pi are a standard USB keyboard and a standard USB mouse. I am using a regular HDMI monitor, no touchscreen.
I can start Ground Control. (Yeah)
However, a round mouse pointer only appears on the main control page when I left click the mouse.
That means I have the mouse clicked to see where I am navigating.
I managed to get into the Settings Menu and confirmed that the mouse is switched to be "on". 
Once I see the buttons of one of the Settings Menus I can't click and drag the mouse anymore as I am then activating the buttons.
Also if I click to long, Ground control aborts and I get kicked out.
Any suggestions how I could fix this issue?
Thanks
Guido

davidelang, 2017-09-10:
the problem is that the default setup is assuming that you have a touchscreen, 
there should be a kirvy setting that switches between touchscreen and mouse mode

guidobender, 2017-09-10:
I open GroundControl as root using sudo.
If I do:
sudo nano /root/.kivy/config.ini
I see:
[input]
mouse = mouse,disable_multitouch
This was the initial setting. It doesn't seem to disable it then it seems. Or is there something incorrect?
Thanks
Guido

guidobender, 2017-09-10:
Maybe I am assuming too much. I thought that when I change the config.ini file and then start groundcontrol, the .ini file is read when starting groundcontrol and the values changes.
Are there any default settings somewhere else? I am pretty much at a dead end and have no idea how to make this work.
Thanks a bunch
Guido

guidobender, 2017-09-11:
This is turning into a monologue here, but I found some other threads that I tried. So far none of them has worked for me.
https://10-009.wikispaces.com/How+to+show+mouse+cursor+for+Kivy+on+Raspberry+Pi%3F

https://stackoverflow.com/questions/43588499/kivy-mouse-cursor-not-showing
If you have other ideas, please let me know.
Thanks
Guido

guidobender, 2017-09-11:
Well, last time for me. It is working now thanks to this thread:
https://stackoverflow.com/questions/45281544/detect-which-mouse-button-was-pressed-in-kivy-on-raspberry-pi

You need to open the .ini file by:
sudo nano /root/.kivy/config.ini

Go to [modules] and enter:
[modules]
touchring = show_cursor=true
cursor = 1

Just in case I also repeated that with the config.ini at
sudo nano ~/.kivy/config.ini

I hope that helps somebody at one point.
Tschöhö
Guido

closed 2017-09-11 by guidobender