MaslowCNC/GroundControl #423: pyserial version for Raspberry Pi

servant74, 2017-09-18:
I keep getting the message 
    Pyserial version 3.x is needed version 2.6 is s installed

How can I solve this issue?

This is on a Raspberry Pi 3 with kivy

GeroBH, 2017-09-19:
You could try last post in http://www.maslowcnc.com/oldforums#!/software-issues:connection-keeps-dropping

BarbourSmith, 2017-09-21:
Were you able to resolve this?

servant74, 2017-09-21:
> <> ... Jack

No I have not.  Thanks for asking Bar... I am starting on the physical
build today.  Finally opened the box this afternoon after getting some wood
this morning.

I have set aside the Pi until I get a bit of the physical build done, but
will be back to it by the weekend, hopefully.

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

BarbourSmith, 2017-09-22:
Have you tried running 

`pip install pyserial --upgrade`

from the command prompt? 

Other than that the only thing I can think to do would be to uninstall pyserial and reinstall it 😬

davidelang, 2017-09-22:
it may also be that you have both python 2.7 and python 3 installed and only 
have pyserial for 2.7 (or something along those lines)

servant74, 2017-09-22:
I issued the pop command Bar suggested.  It added 2.6 version while noting
that 3.x (something, sorry I missed the exact version) was there too.

After that starting Ground Control seems to work.

... There are various ports listed under (Actions -> Ports) ... any guess
how to choose the appropriate one?  Otherwise the Monte Carlo method will
be invoked. <grin>

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

davidelang, 2017-09-22:
one of them will appear and disappear as you plug and unplug the maslow. That's 
the one to use.

k

BarbourSmith, 2017-09-22:
@davidelang  is right about the best way to find the right port (you only need to do it once, then it will remember). It will also be the same one as in the Arduino program.

I'm going to close the issue 😁

closed 2017-09-22 by BarbourSmith