MaslowCNC/GroundControl #424: Make Ground Control launch full screen again

BarbourSmith, 2017-09-24:
I liked it better when Ground Control launched full screen. Is anyone opposed to bringing that back? It can always be reduced in size using the [image] in the upper right corner

blurfl, 2017-09-24:
Hadn’t commented, but I agree - full screen was better. The Notification popups all come up collapsed with the current windowed configuration :( 
👍

krkeegan, 2017-09-25:
I am also in favor of returning to full screen
:+1:

closed 2017-09-25 by BarbourSmith

servant74, 2017-09-25:
> <> ... Jack

Ok.  How about either a command line switch, -f or -w (full or windowed) or
similar to allow the other option to be specified on the command line when
launching. ... Just a thought.

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

blurfl, 2017-09-25:
>How about either a command line switch, -f or -w (full or windowed) or
similar

I can see how to pass an argument into main.py and choose whether to execute the Window.maximize based on that. The argument must not be accepted by kivy or python, so -w or --windowed won't work. 'w' or 'windowed' would do, though:

        arg = sys.argv[1].lower() if len(sys.argv) > 1 else 'h'
        if ((arg != 'w') & (arg != 'windowed')):
            Window.maximize()

Would starting from the command line with 'kivy main.py windowed' achieve the goal?

servant74, 2017-09-25:
Sure.  For use with kivy on a pi with the 7" touch screen, starting full
screen seems like the most appropriate option to me.
Windows or Mac use might be different.

Being from an old programmer background, I found giving alternatives that
users can implement with relative ease, cuts down the trouble tickets in
the long run, especially if the documentation users read jives well with
actual use.

No real issue for me, it is just my old programmer voice that keeps coming
out after 40 years in the trenches (and glad to be retired!).

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

blurfl, 2017-09-25:
Are you using a Pi with the touch screen? How do you find GroundControl runs on that screen? 

Unless there’s more interest in reviving the windowed mode, let’s let this dog sleep. Each OS provides ways to for the user manage the window size, we can rely on that.

servant74, 2017-09-25:
Yes, that is what I have, the Pi 3 and the 7" touch screen, and the pi
touch case for them too.  It seems to run pretty well.  I don't have my
machine running, but will keep you updated if you like.
It seemed to be a 'natural paring' for me, especially for a garage unit.  I
dd break down and get a small keyboard to put in file names, but that is
the worst as far as I can tell.
Also being an old programmer/sysadmin from back when, and loving
toys/tools, I don't hesitate to long to try something, but as I have gotten
older, I do enjoy a bit of hand holding when issues arise.

If you don't think adding that option would add value for others, I am
willing to help keep down 'creeping featurism'.

Thanks for the help and support... Jack

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

PR #427 (2017-09-25): Maximize GC on launch
#418 brought a lot of great changes to help things render better in small windows, but it seems like we probably want GC to launch maximized as described in #424 .

This PR puts back in the line to auto maximize GC and fixes #424