MaslowCNC/GroundControl #425: Test cut problems

benkeller3, 2017-09-24:
BenKeller3
As I understand this calibration process … you should run a test cut … measure the result and load it and then rerun the test cut and the two numbers should come together gradually.

Mine bounce around but never get close.

1st H 602 V 593
2nd 601.5 594
3rd 601.5 593
4th 602 592

This can not be right … I have re-calibrated everything else .

I am lost on this one.

Also running a 45 degree pantograph linkage design but that would not seem to matter

I changed to Triangle and measured and input the end of chain to center of the router bit in advanced settings.

blurfl4m
I wonder if the calibration routine accommodates the triangular Kinematics? The test cut routine adjusts the distance between sled mounting points for the original setup, not sure what it would adjust for the triangular one.
This sounds like it would be good to open as a new issue in GroundControl. Would you do that?
With all the recent interest in alternative sled connections, we may well more of see this soon.

davidelang, 2017-09-24:
it would need to adjust the motor spacing dimension, but that should be measured 
pretty accurately (assuming that the frame doesn't flex as the chain is drawn 
tight)

krkeegan, 2017-09-25:
I did this manually.  I cut horizontal and vertically spaced marks and measured the distance.  I adjusted the rotation radius until the distances matched.  I figured whatever error I had was caused by that.  

Was that a bad idea, should I have been tweaking the motor spacing?

davidelang, 2017-09-25:
On Mon, 25 Sep 2017, Kevin Robert Keegan wrote:

> I did this manually.  I cut horizontal and vertically spaced marks and measured the distance.  I adjusted the rotation radius until the distances matched.  I figured whatever error I had was caused by that.
>
> Was that a bad idea, should I have been tweaking the motor spacing?

play with the simulator. If the rotation radius is wrong, it will show up the 
same as a chain length error.

I suspect that motor spacing errors will look very similar, but I haven't 
checked that.

benkeller3, 2017-09-25:
So in doing some research on how the calibration with test cut works now (it is adjusting the sledWidth)  can I not take the measureMachinePopup.py (the calibration portion) and change the it to adjust the the motorSpacingX  seems logical 

Any thoughts

Here is the code section .

  amtToChange = .9*dif
            newSledSpacing = float(self.data.config.get('Maslow Settings', 'sledWidth')) + amtToChange
            print "Now trying spacing: " + str(newSledSpacing)
            self.data.config.set('Maslow Settings', 'sledWidth', str(newSledSpacing))
            self.data.config.write()
            self.cutBtn.disabled = False
            self.data.pushSettings()

BarbourSmith, 2017-09-25:
From playing with the simulator an error in either the ring radius or the motor spacing causes similar distortions. 

I was hoping that the rotation radius would be something fixed once we have a standard kit available, but it looks like there are going to be a number of options out there so we can't count on that number being known. If we knew the radius for sure it would make sense to adjust the motor spacing. If we don't have a good measurement of the radius it probably makes sense to adjust that.

I think we will also need to change the test pattern cut, right?

Does anyone have thoughts on a test pattern which would work well?

I've been using three horizontal marks near the top to check for any curve in straight lines, and two vertical marks spaced far apart near the bottom to test for any error in the length of long parts.

Here are my test files:
[Test Shapes.zip](https://github.com/MaslowCNC/GroundControl/files/1330296/Test.Shapes.zip)

krkeegan, 2017-09-25:
I had been using a square shape and comparing the width and height.

@BarbourSmith if I understand your process correctly, the compulsively accurate side of me worries about using the edge of the work surface to check for straight lines.  It may not be level, and I have certainly seen unsquare plywood (although it is rare).  If you are using a straight edge I found that even over long distances it is really hard to detect a curve, but it was easier to measure the distance between two points.

BarbourSmith, 2017-09-25:
I completely agree about using the plywood as a straight edge is a questionable idea. I've been making the three marks maybe 8 inches from the top and then using a straight edge to check them.

I agree that measuring the distance between two points is simpler and easier, and from my experience the two tests are redundant. If one looks good, then the other will too. 

From playing with the simulation it seems to me like two points a known (long) distance apart helps dial in the triangular calibration more quickly than a square, but having just two points means we need to take into account the width of the bit

blurfl, 2017-09-25:
3> three marks maybe 8 inches from the top and then using a straight edge to check them 

Stretching dental floss between pushpins at the endpoints makes a cheap, accurate straight edge.

> having just two points... the width of the bit

The width of the bit is eliminated if we measure from the one side to the same side, or do I misunderstand the measurement?

BarbourSmith, 2017-09-25:
@blurfl Great point about the width of the bit not mattering as long as we measure from the same point on each cut! We've just got to make sure the instructions are clear and I think that will work!

davidelang, 2017-09-25:
On Mon, 25 Sep 2017, BarbourSmith wrote:

> From playing with the simulator an error in either the ring radius or the motor spacing causes similar distortions.

that's what I was guessing would happen

> I was hoping that the rotation radius would be something fixed once we have a
> standard kit available, but it looks like there are going to be a number of
> options out there so we can't count on [...]

It's far easier to measure the rotation radius than it is to measure the motor 
spacing. If they both have similar effects, can you cancel the error from one 
out with deliberate errors on the other?

> I think we will also need to change the test pattern cut, right?

We should make the test pattern be whatever makes it easiest to detect errors. 
If errors happen in one direction consistantly enough that once you work those 
errors out, the errors in the other direction are gone, then we only need to 
have the pattern show errors in one direction.

> Does anyone have thoughts on a test pattern which would work well?

I need to fiddle with the simulator to figure out what the errors look like.

davidelang, 2017-09-25:
On Mon, 25 Sep 2017, BarbourSmith wrote:

> From playing with the simulation it seems to me like two points a known (long) 
> distance apart helps dial in the triangular calibration more quickly than a 
> square, but having just two points means we need to take into account the 
> width of the bit

measure from the right side of one slot to the right slide of the other slot, 
not right to left.

Remember that people are going to be measuring this with a tape measure, so that 
limits your accuracy to ~1mm at best.

benkeller3, 2017-09-26:
So we have a measurement process and we will adjust the  motor spacing
dimension but is there any science to this or is it back to my Artillery
days and shoot a test cut  ....measure adjust the motor distance  and
 shoot another test .    if it went the wrong way correct it it closed
interpolate a new adjustment and walk your way into the target of perfect
alignment...

(a true artillery guy is going to cringe because we were supposed to
bracket the target not walk it in...<grin>)

blurfl, 2017-09-26:
The quadrilateral auto-calibration process uses this approach, and on a stiff frame it gets to within the design accuracy in three or four iterations. No reason to think the triangular couldn’t do as well.
If we know the distance between the motors, shouldn’t we be able to calculate the distance between a sprocket and the router bit by cutting two horizontal marks on the vertical centerline and measuring the actual distance between?

blurfl, 2017-09-26:
Another approach would be to have the user measure and mark the vertical centerpoint of the workarea, then programmatically move the sled to the position determined by ORIGINCHAINLEN. The user would then use up/down arrow arrows to move the bit to the measured vertical center. The difference between that chain measurement at that point and ORIGINCHAINLEN would be proportional to the distance between the bit and the end of the chain.

BarbourSmith, 2018-02-01:
I believe that this has been resolved with the new calibration process! 🎉

closed 2018-02-01 by BarbourSmith