MaslowCNC/GroundControl #428: Gcode post 6000 lines cutting random places

jbnimble, 2017-09-28:
As noted in the [forum](http://forums.maslowcnc.com/t/gcode-post-6000-lines-cutting-random-places/737). I had an issue where the cut went a little strange after approximately 7000 Gcode lines were executed. Here are the files:

These are the two log files created during the day of cutting, not sure which one is for when.
[kivy_17-09-26_1.txt](https://github.com/MaslowCNC/GroundControl/files/1339125/kivy_17-09-26_1.txt)
[kivy_17-09-26_0.txt](https://github.com/MaslowCNC/GroundControl/files/1339124/kivy_17-09-26_0.txt)
Github does not allow .nc files, so this is the GCode file as a TXT.
[lilyspod-8th.txt](https://github.com/MaslowCNC/GroundControl/files/1339128/lilyspod-8th.txt)
The image, put some squares around the areas where things started going sideways.
[image]

blurfl, 2017-10-16:
This looks like it might be the same issue as #428 in Firmware - space runs in the gcode lines. GroundControl PR#434 addresses that issue. @jbnimble, could you try it to see if it fixes the problem you were having?

jbnimble, 2017-10-24:
Running v0.94 and the moment 60000 lines hit the sled started heading off in a SE direction. Here is the file. This is a slightly different file because I did not want to be cutting for a lot of hours, but it went sideways right at 60000 lines executed. The Hold button did not work, I had to Stop the execution all together.

[60000plus_gcode.txt](https://github.com/MaslowCNC/GroundControl/files/1409149/60000plus_gcode.txt)

blurfl, 2017-10-24:
Can you post the log.txt file? It should show the commands sent.

blurfl, 2017-10-24:
@BarbourSmith, can you think of a way to modify a copy of firmware so it runs without moving the motors? Sure would be useful to try to replicate this if one didn't have to wait for the motors to move😀

jbnimble, 2017-10-24:
Latest log files for most recent cut.
[kivy_17-10-23_0.txt](https://github.com/MaslowCNC/GroundControl/files/1409234/kivy_17-10-23_0.txt)
[kivy_17-10-23_1.txt](https://github.com/MaslowCNC/GroundControl/files/1409235/kivy_17-10-23_1.txt)
[kivy_17-10-23_2.txt](https://github.com/MaslowCNC/GroundControl/files/1409236/kivy_17-10-23_2.txt)
[kivy_17-10-23_3.txt](https://github.com/MaslowCNC/GroundControl/files/1409237/kivy_17-10-23_3.txt)
[kivy_17-10-23_4.txt](https://github.com/MaslowCNC/GroundControl/files/1409238/kivy_17-10-23_4.txt)
[kivy_17-10-23_5.txt](https://github.com/MaslowCNC/GroundControl/files/1409239/kivy_17-10-23_5.txt)
[kivy_17-10-23_6.txt](https://github.com/MaslowCNC/GroundControl/files/1409232/kivy_17-10-23_6.txt)
[kivy_17-10-23_7.txt](https://github.com/MaslowCNC/GroundControl/files/1409233/kivy_17-10-23_7.txt)
[kivy_17-10-23_8.txt](https://github.com/MaslowCNC/GroundControl/files/1409228/kivy_17-10-23_8.txt)
[kivy_17-10-23_9.txt](https://github.com/MaslowCNC/GroundControl/files/1409229/kivy_17-10-23_9.txt)

I imagine the 9th one is the important one, but those are all my logs from today.

blurfl, 2017-10-24:
Not sure which platform you're using, but in the same folder as main.py there will be a file log.txt which is the capture of all the serial comm between GC and the firmware.
 In Linux it's in the GroundControlv0.nn folder.
 On the Mac it will be in:
/Applications/groundcontrol.app/Contents/Resources/yourapp/log.txt
In Windows, its... well, in the same place as main.py and groundcontrol.py.

That is what I was hoping to see; it will be be _big_, so it may need to be zipped.

jbnimble, 2017-10-24:
Log file.
[log.txt](https://github.com/MaslowCNC/GroundControl/files/1410502/log.txt)

blurfl, 2017-10-24:
Thanks for the log file.
@BarbourSmith, I can see that the entire list of gcodes is present in self.data.gcode, and serialPortThread _write doesn't care about the number of elements in self.data.gcode. Could this be caused by the routine that renders the present location during cutting?

BarbourSmith, 2017-10-24:
I'm not sure I understand the question but I'm going to try to answer it anyway 😀 

self.data.gcode should contain the whole file regardless of how big it is. The GcodeCanvas object (unrelated to cutting parts) will only render the first 60,000 lines because if your computer has a slow graphics card (like mine) more than that can make GC so slow it's pretty much unusable. serialPortThread._write() doesn't care about the length of the file, it just keeps sending the next line until there are no more.

Basically I agree with everything you said.

The fact that the happens right at 60,000 lines doesn't seem like a coincidence. 

What if we make the number of lines to render a user defined setting? That way we could easily change it and see if the issue keeps happening or goes away

blurfl, 2017-10-24:
Making it a user setting would be a good work-around. I was trying to figure out where the error arose. Shouldn't be from serialPortThread, because it doesn't know about 60000. I thought that the routine that renders the current actual position in red might be a place to look. Is that in gcodeCanvas.py? I haven't found it yet.

BarbourSmith, 2017-10-25:
Oh, that one is a little hidden. It's in positionIndicator.py gcodeCanvas has two instances of positionIndicator one for each color

blurfl, 2017-10-26:
I set the limit to 250,000 lines and was able to read in the whole of the lilyspod-8th.txt file and render it on an OSX system.
 I modified a copy of the firmware to bypass the motor commands, shortened the _watchdog time by 90%, so the firmware would receive gcode commands and reply to just move the cursor. I compiled it to run on a 120Mhz Teensy3.5 with the baud rate set to 500,000 (yes, _really_!) and ran it against that file for four hours, watching the red circle wander along the very complex pattern. Got to around 70% done before deciding to wrap it up. GC and the firmware seem stable, though large parts of the firmware were bypassed for this. GC's rendering and the serial communication link seem good at any rate.
 I didn't see whether the cursor went off into space - my eyes were crossed by the end. 😝 Another test would be to run the same firmware on the Mega; maybe its more limited resources are part of the issue. A software tool to validate the gcode lines returned against the lines sent would be interesting to use and challenging to create - several lines can be sent between any one line and its matching response, and whole blocks of gcode are repeated with different z depths
 I wish I had found something more interesting. Keep on digging.

blurfl, 2018-02-18:
I think I've seen some light on this one. If 'Home' is not moved, the cut runs to completion normally. I see the issue when I've moved the 'Home' position, and the cut reaches the 501st line past the limit of lines rendered (60000 presently).  I describe below how to reliably cause the issue to happen, using the file linked [above](https://github.com/MaslowCNC/GroundControl/files/1409149/60000plus_gcode.txt).

 First, to see proper operation, load the file, make sure  'Home' is at 0,0, use 'GoTo' to queue up line 60480. When run, the target will move to the lower right corner and begin cutting where the letter 'a' would be in the name 'Linda'. Watch the line numbers go by past line 60501 and note the target's general location properly within the cut. Make a note of the X andY coordinates in general, they will be in the neighborhood of X511.1024 Y-500.8103. If left running, the cut will finish properly (the last cuts after the 'a' are in the tail of the 'J').

 Next, to see the error occur, set 'Home'  to X-350.0mm Y350.0mm. I choose this point both because it will exaggerate the error when it happens, and because I imagine that @jbnimble probably had done something like that to put the cut pattern in the middle of the workarea. Again use 'GoTo' to queue up line 60480 and watch the target. When it executes line 60501 it will travel 'out of bounds' and the gcode for that line will show that the coordinates are not based on our moved 'Home' but on 0,0 (have jumped from the range of X161.0378 Y-150.8242 to the range near X511.1024 Y-500.8103 starting with line 60501) .

 Somehow, the coordinate frame is getting lost. gcodeCanvas::callBackMechanism() works in 500 line chunks, that seems like a place to begin looking.

BarbourSmith, 2018-02-19:
Yes yes yes!!!!

This has been at the back of my mind forever because we never found a solution and I couldn't reproduce it, but it was reported a couple times so I knew it was out there.

Thank you for finding a way to replicate it, and thank you for sticking with it...the early indications seem like something strange is going on, I'm excited to find out what it is 😀