MaslowCNC/GroundControl #431: Updated to ver 91, but Ground Control on Raspberry pi says it is 92

servant74, 2017-09-28:
Version 91 announcement came out yesterday.  I updated the firmware and GC, but GC reports ver 92, and firmware ver 91.

I would correct, but not sure how.

GroundControl for RaspberryPi

krkeegan, 2017-09-28:
What is your process for getting your GC version?  Do you use git, if so what is the command, or do you download from the releases (https://github.com/MaslowCNC/GroundControl/releases) page?

servant74, 2017-09-29:
> <> ... Jack

I use the following command

  git clone https://www.github.com/MaslowCNC/GroundControl.git

-- 

The Four Boxes of Liberty - "There are four boxes to be used in the defense
of liberty: soap, ballot, jury and ammo. Please use in that order."
"Whatever you do, work at it with all your heart"... Colossians 3:23
"Anyone who has never made a mistake, has never tried anything new." -
Albert Einstein
"You don't manage people; you manage things. You lead people." - Admiral
Grace Hopper, USN
"The most dangerous phrase in the language is "We’ve always done it this
way"-- Admiral Grace Hopper, USN
"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

krkeegan, 2017-09-29:
That particular command is how you initialize the git reposititory.  But a fetch or pull will always get the most recent version, not necessarily the released version.  In this case, commits have already been made since the release on Wednesday, including one which has bumped the version number.

To get a specific release use:

git checkout v0.91

Alternatively you can just download the release zip files as well:

https://github.com/MaslowCNC/GroundControl/releases
https://github.com/MaslowCNC/Firmware/releases

BarbourSmith, 2017-09-29:
@krkeegan is spot on, the "most recent" version should always be one version ahead of the last release. The git pull command will usually get you a version which will work, but keep in mind that you are running the cutting edge version which is updated every day, not the release which is in theory a little more stable and tested

servant74, 2017-09-29:
> @krkeegan <https://github.com/krkeegan> is spot on, the "most recent"
> version should always be one version ahead of the last release. The git
> pull command will usually get you a version which will [...]

Great, thanks for the help.  I did not know that.
Please mark this issue closed.  Once I try it out I will update the wiki
for Raspberry Pi ...

BarbourSmith, 2017-09-29:
Thanks for updating the wiki to help others in the future. Your rock! 👍 👍 👍

closed 2017-09-29 by BarbourSmith