MaslowCNC/GroundControl #444: Calibration process improvements

BarbourSmith, 2017-11-02:
I just went through the calibration process for the triangular kinematics and had some ideas for how to make it better.

1) Have an option to enter a guess for the radius of rotation. Right now it just assumes you are using @pillageTHENburn's linkage kit so it takes a lot of tries to end up in the right spot

2) The cut marks move steadily up and to the right with each iteration. They should move up, but zigzag to keep the line roughly centered

davidelang, 2017-11-03:
Please look at my thread on the topic 
https://forums.maslowcnc.com/t/calibration-pattern-for-triangular-kinematics/879/

for triangular kinematics, we should be able to calculate the correct dimensions 
after one cut.

krkeegan, 2017-12-16:
I agree with point 1 which I think still needs to be fixed

Point 2 seems to have a slider now?  I would say at least the slider shouldn't default to full bottom which it did for me, as that causes my sled to fall onto the floor.  Perhaps even make it so that users can't cut that low down on the frame even?  Since we can't actually see the test cut placements.

Also getting to .5MM at this distance is way hard for me.  I use an inches tape measure, which is probably no better than 1/16 accuracy, I suspect most ppl will as well.  I eventually gave up aroubd 1/32 of error that seemed to be bouncing back and forth and just lied and told it 75 inches.

davidelang, 2017-12-16:
If we are measuring things <6" or so, the answer is to go to harbor freight and 
spend $10 to buy an electronic caliper

But the other approach is to try to make the baseline as large as possible so 
that the measurement inaccuracy is a smaller percentage of the baseline.

That's why in the triangular kinematics suggestion, I try to use as much of the 
vertical space as possible.

also keep in mind that people are building differenet size machines, some taller 
(to handle 5'x5' plywood) and some smaller. So the calibration should not assume 
standard machine dimensions. This is why in my triangular calibration 
suggestion, I have the user move the sled to the middle, and then go from there.

blurfl, 2018-01-14:
Much of this has been addressed in PRs - can this be closed?

BarbourSmith, 2018-01-15:
I agree. The features I mentioned have been replaced by others. I will close.

closed 2018-01-15 by BarbourSmith