MaslowCNC/GroundControl #446: Linux install

Avow-Truth, 2017-11-06:
I'm a bit familiar with clicking my way through Linux. And I've used command prompt and power shell a few times so working in terminal isn't completely new. (But most of it is still unnatural to me)

So I'm at a loss what to do after doing the install steps and executing the "python main.py" in terminal I get this back. 

`:/media/ben/HOME/user/GroundControl-master$` python main.py
[INFO   ] [Logger      ] Record log in /home/ben/.kivy/logs/kivy_17-11-06_2.txt
[INFO   ] [Kivy        ] v1.10.0
[INFO   ] [Python      ] v2.7.14 (default, Sep 23 2017, 22:06:14) 
[GCC 7.2.0]
 Traceback (most recent call last):
   File "main.py", line 11, in <module>
     from kivy.app                   import App
   File "/usr/lib/python2.7/dist-packages/kivy/app.py", line 319, in <module>
     from kivy.base import runTouchApp, stopTouchApp
   File "/usr/lib/python2.7/dist-packages/kivy/base.py", line 30, in <module>
     from kivy.event import EventDispatcher
   File "/usr/lib/python2.7/dist-packages/kivy/event.py", line 8, in <module>
     import kivy._event
 ImportError: /usr/lib/python2.7/dist-packages/kivy/_event.i386-linux-gnu.so: undefined symbol: PyFPE_jbuf
:/media/ben/HOME/user/GroundControl-master$`

I'm guessing something didn't take on the install with kivy?

jbnimble, 2017-11-06:
Had a similar issue, basically the solution was to install Kivy from source, and add some missing packages, the [forum thread I posted is here](https://forums.maslowcnc.com/t/kivy-broken-in-ubuntu-17-10/885).

Avow-Truth, 2017-11-06:
Ok thanks for the quick reply. I'll have to try that later tonight when I'm off.

krkeegan, 2017-12-20:
Did installing from source fix you issue?  Can we close this?

closed 2018-01-10 by BarbourSmith