MaslowCNC/GroundControl #448: Auto calibration - Step 7 suggestion

bdillahu, 2017-11-09:
On step 7, where it pays out first the left chain, then the right preparatory to connect to the sled, suggest a few seconds between sides, or a "next prompt" of some kind to make sure you can get from one side to the other. I'm holding one chain to keep it from tangling, and now that other one's moving and I've got to get this one laid somewhere and get over there :-)

blurfl, 2017-11-12:
I second the motion. I dislike having to stand by the right motor waiting. Suggest a 'next' click followed by a 5 second pause to get from the mouse to the right motor 🙂

BarbourSmith, 2017-11-13:
I'm on the other side. I would never encourage anyone else to, but I usually let my calibration process run out both chains without me touching anything. I've found that as long as the first link isn't too well planted on the gear they won't wrap (but I would never encourage anyone to do the same).

What about a solution that gives us both options:

We could have two buttons one for "measure out left chain" and one for "measure out right chain". Clicking either one would start a 5 second count down followed by the chain extending. If you want the current behavior where both chains extend automatically you could just click one, then the other and it would queue up the second one to start right after the first finished.

How do we feel about that as a solution? I think I can implement that pretty easily.

bdillahu, 2017-11-13:
Works for me... 

I know in my layout it isn't the chain getting tangled at the sprocket, but rather farther down as it falls it gets caught on itself headed back up... Probably varies a good bit depending on people's frames and the angles involved.

blurfl, 2017-11-13:
"farther down as it falls" I've had that too. I've been thinking of a piece of 'stretchy string' with a hook or magnet on the end to manage the chain while it feeds in the various calibration steps.

bdillahu, 2017-11-13:
That’s a good idea...

commit 8c448403 (2017-11-29): Add one button for extending each chain

Adds one button for extending each chain to the calibration process.

Fixes #448
PR #479 (2017-11-29): Add one button for extending each chain
Adds one button for extending each chain to the calibration process.

Fixes #448
closed 2017-11-29 by BarbourSmith