MaslowCNC/GroundControl #451: Clicking title bar initiates mouse panning on OSX

skiz, 2017-11-11:
On mouse up, when clicking on the main window bar to move the window, the routine for panning the view is initiated, and can only be stopped by clicking inside the window.

Installed from source per instructions @ commit 348ca07a48cb02d7b3c98ed4d01a53798fea3b63

GroundControl$ python main.py
[WARNING] [Config      ] Older configuration version detected (0 instead of 19)
[WARNING] [Config      ] Upgrading configuration in progress.
[INFO   ] [Logger      ] Record log in /Users/jmartin/.kivy/logs/kivy_17-11-11_0.txt
[INFO   ] [Kivy        ] v1.10.0
[INFO   ] [Python      ] v2.7.13 (default, Mar 14 2017, 00:54:17) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.42.1)]
[INFO   ] [Factory     ] 194 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_imageio, img_dds, img_sdl2, img_gif (img_pil, img_ffpyplayer ignored)
[INFO   ] [OSC         ] using <multiprocessing> for socket
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <2.1 INTEL-10.25.13>
[INFO   ] [GL          ] OpenGL vendor <Intel Inc.>
[INFO   ] [GL          ] OpenGL renderer <Intel HD Graphics 4000 OpenGL Engine>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <1.20>
[INFO   ] [GL          ] Texture max size <16384>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [Base        ] Start application main loop
[INFO   ] [GL          ] NPOT texture support is available

BarbourSmith, 2017-11-13:
Very interesting. I don't see the same behavior on Windows

This almost feels like a bug in Kivy (our window provider), right? That doesn't mean we shouldn't fix it.

Would the right fix be to check for the click location and not accept it if we are in the top of the frame?

skiz, 2017-11-18:
I attempted to dig in to this last night, and I'm also thinking that it is Kivy from  the investigation.  I'm trying to reproduce it in isolation but haven't been able to yet.  I also haven't seen any bug reports on Kivy itself for this.

BarbourSmith, 2018-01-10:
The buttons showing off the top issue has been fixed...do we feel good about closing this?

blurfl, 2018-01-10:
Close it, it's fixed.

closed 2018-01-10 by BarbourSmith