MaslowCNC/GroundControl #455: Setup Machine Dimensions - Step 10 of 10

jbnimble, 2017-11-17:
After selecting Stop Cut there is no way to restart the cut. If for example you realize the first cut did not go deep enough because the Z-Axis zeroing was incorrect, and you go back to fix the Z-Axis, but now you can't start the cut over again.

It would be nice to be able to either send the sled to Home, or just restart the cut from its current location.

BarbourSmith, 2017-11-17:
Great point!

What if I made it so that it just re-enabled the "Run Test Cut" button after stop is pressed so that pressing that button would restart it

jbnimble, 2017-11-17:
That would be fine with me, also having it start the test cut over again from the start would be fine.

BarbourSmith, 2018-01-10:
> What if I made it so that it just re-enabled the "Run Test Cut" button after stop is pressed so that pressing that button would restart it

This has been added! Excellent suggestion 👍 👍

closed 2018-01-10 by BarbourSmith