MaslowCNC/GroundControl #456: Perform Test Cut Anytime (or after automatic chain calibration)

jbnimble, 2017-11-17:
It would be convenient to be able to optionally perform the test cut procedure after the automatic chain length calibration. Instead of having to go through the whole 10 step calibration process to get to the point of being able to perform the test cut.

Even better it would be convenient to be able to perform the test cut at any time.

BarbourSmith, 2017-11-17:
That is a great idea. Let me think on how to best impliment that

BarbourSmith, 2017-11-20:
Great recommendation @skiz I will add a test gcode file to the repo

krkeegan, 2017-12-18:
I have been able to skip through the calibration steps to the test cut and redo it that way.  It seems to work just fine as long as you use the skip button nothing gets erased or changed in the other steps.

BarbourSmith, 2017-12-18:
I just broke the test cut out into it's own object so it should be pretty easy now to put it in a popup up under something like "Actions -> Perform Triangular Test Cut" or something similar so we can run it as a stand alone.

I agree with @krkeegan that using only the 'skip' button should work too