MaslowCNC/GroundControl #460: Actions -> Calibrate Chains Length - Automatic does not finish at 0,0

blurfl, 2017-11-18:
It looks like though logic to move to 0,0 at the end of chain calibration in the full calibration sequence, it wasn't added to the end of the the automatic calibration on the Actions page. See discussion [here](https://forums.maslowcnc.com/t/i-put-my-linkage-kit-together/854/42?u=blurfl).

BarbourSmith, 2017-11-20:
I think this is part of a larger issue that we have the same code implemented twice. Really there should be a single instance that is referenced in both places, right? That way a change to one will show up in both places

blurfl, 2017-11-20:
You're right, that's always the best. Got to be able to use the Automatic Chain calibration without entering the full calibration sequence, though.

BarbourSmith, 2017-11-29:
Fixed in #477 👍

closed 2017-11-29 by BarbourSmith