MaslowCNC/GroundControl #465: Triangular Calibration Drops Sled to Floor

madgrizzle, 2017-11-20:
Reference: [https://forums.maslowcnc.com/t/calibration-linkage-alternate-sled-design/1093/12](url)

I looked at the code and didn't see any glaring mistakes, but the sled drops to the lower left corner during the final step of calibration and if using the temporary frame, rolls around on the floor.

Perhaps X0, Y0 that's determined during the calibration process is off.  I didn't readily see anything in the calibration code that would cause it to drop so low.

BarbourSmith, 2017-11-20:
I'm looking into this right now

madgrizzle, 2017-11-20:
Isn't the kinematics settings under advanced settings not maslow settings:

This line is from the calibration routine:
self.data.config.set('Maslow Settings', 'kinematicsType', self.chooseKinematicsType.text)

BarbourSmith, 2017-11-20:
I've made this change and some other small fixes. When I ran the calibration process on a fresh install of Ground Control (deleting the .ini file in your home directory to wipe all the settings to the default) it didn't go off the bottom for me.

When you have a chance will you try the latest version? I'm happy to compile one for you if you aren't set up to run python directly

PR #481 (2017-11-29): Add slider to move test cut up and down
Adds a slider to allow you to move where the test pattern is cut on the sheet.

Fixes #465
closed 2017-11-29 by BarbourSmith