MaslowCNC/GroundControl #466: it's not clear that exiting the calibration process part way through can cause trouble

blurfl, 2017-11-20:
Some steps of the auto calibration process put the software into relative mode, or alter the previous settings. Advancement through the calibration steps in other than the default order can result in lost or improper settings.

skiz, 2017-12-12:
Would it make sense to restore the original settings on exit unless they explicitly save changes from the last page?   showing current & new values could be beneficial, but the screen is already quite crowded on a pi 7"

blurfl, 2017-12-12:
I agree on both items. 👍🏻

madgrizzle, 2018-01-05:
This could be the cause of some of the strange issues I had during calibration early on.  I had many failed attempts and if I aborted or skipped some steps, sometimes it behaved weirdly.  I suggest that perhaps each step of the calibration issue a command to ensure the controller is in its proper (relative or absolute) mode, regardless of the mode of the previous step.

blurfl, 2018-01-09:
I think the issue I was seeing was caused by the 'Begin' button in Step1 - it executes gcode "B06 L0 R0" which tells both axes that they are at the first link of the chain. There is a comment "This prevents strange behavior when rotating each sprocket to 12:00". 
 With the present code, clicking this button clobbers your chain calibration whether that was what you intended or not.
I'll do some testing, but I believe recent changes to kinematics and the chain calibration procedure make this 'Begin' button behaviour unnecessary.

BarbourSmith, 2018-01-09:
I agree that we've made significant improvements to this area and it's probably safe to remove that command. It was in there to prevent issues which were showing up when the axis target and actual location were far apart at the beginning of the calibration process which is a state which could be left from a previous failed calibration attempt. 

We are much less likely to end up in that situation now so I say go for it and take it out!

BarbourSmith, 2018-01-10:
@blurfl , is this issue good to close or is more needed do you think?

blurfl, 2018-01-10:
I'd say it's done.

closed 2018-01-10 by BarbourSmith