MaslowCNC/GroundControl #482: Triangular calibration uses default value instead of previous value

blurfl, 2017-11-30:
Trying to do the calibration step with a top-pantograph linkage (129mm rotationradius), I find that choosing the Triangular calibration writes 260 to rotationradius in groundcontrol.ini overwriting the previous value. That should not happen.

 Further, couldn't we check and use the previous value if it is present? That, and allow the user to input an approximate value when choosing Triangular similar to the step in Quad. that makes an estimate based on chain links.

BarbourSmith, 2017-11-30:
Yes! This is something I've been thinking about. I think I'm going to remove the step where you count the links and put a new step after choosing the kinematics type which asks for a guess at the radius or distance between attachment points.

The work around is that as is the calibration process just takes more tries to converge because the initial guess isn't right, but it will get there

blurfl, 2017-11-30:
I've got a PR ready that looks for an existing value and uses that... I see that if groundcontrol.ini doesn'ty have a value for rotationRadius it puts 100 there.

BarbourSmith, 2017-11-30:
Love it. That's already a huge step forwards, let's do it!

blurfl, 2017-12-01:
Darn , I really didn't mean to open this branch on the upstream master 😧! What best to do?

blurfl, 2017-12-02:
Fixed PR#484

closed 2017-12-02 by blurfl