MaslowCNC/GroundControl #486: after PR#485, groundcontrol fails to run

blurfl, 2017-12-01:
after PR#485, groundcontrol fails to run, posts thew following error:

` Traceback (most recent call last):
   File "main.py", line 38, in <module>
     from UIElements.diagnosticsMenu   import   Diagnostics
   File "/Users/scottsmith/Documents/GitHub/MaslowCNC/GroundControl/UIElements/diagnosticsMenu.py", line 5, in <module>
     from UIElements.measureMachinePopup              import    MeasureMachinePopup
   File "/Users/scottsmith/Documents/GitHub/MaslowCNC/GroundControl/UIElements/measureMachinePopup.py", line 338
     except:
           ^
 IndentationError: unindent does not match any outer indentation level
`

blurfl, 2017-12-01:
spacing got changed on lines 334-337...

BarbourSmith, 2017-12-01:
Great catch. The PR was all good, but somehow the merge broke things. I will fix ASAP

blurfl, 2017-12-01:
👍💯

BarbourSmith, 2017-12-01:
We had a rogue space running loose! 😁

closed 2017-12-01 by BarbourSmith

krkeegan, 2017-12-01:
odd, sorry about that guys.  Not sure how that happened.

BarbourSmith, 2017-12-01:
It definitely wasn't your fault, the pull request itself worked great. It think we both edited that section of code and the automatic merge process decided to add an extra space for some reason 🤷‍♂️

krkeegan, 2017-12-01:
odd, never seen anything like that before, I guess it keeps you on your toes.

blurfl, 2017-12-01:
Thanks again for the performance enhancers 👍

PR #487 (2017-12-01): Fix rogue space crashing GC
Somehow added during the merge process 😬 

Fixes #486
commit 0ee3de47 (2017-12-01): Fix rogue space crashing GC

Somehow added during the merge process :grimacing:

Fixes #486