MaslowCNC/GroundControl #490: Add Stop Button to Z-Axis Control Menu

krkeegan, 2017-12-05:
Pretty straight forward, we need a stop button in the zaxis control menu.  If you enter a bad distance which would break your zaxis plastic clip.  You have to quickly exit the zaxis control menu and then hit stop.  A stop button in the menu would be easier, it just needs to send the ! command.

blurfl, 2017-12-06:
This introduces an odd number of buttons to the panel. Any opinions on the best way to lay this out? 4-wide by 2, with one button blank? 
[image]

krkeegan, 2017-12-06:
Just a thought what about stop as the far right button and can it span both rows?

blurfl, 2017-12-06:
I don’t know how to make a button span, I was hoping you did 😄

BarbourSmith, 2017-12-06:
Great suggestion! I know how to do that 😀 

The key is going to be to put it in a second GridLayout and use the size_hint_x variable to make one column be .75 and the other .25 because one column is three buttons wide and the other is one button wide

blurfl, 2017-12-07:
Sort of like this:
[image]

The button would trigger this in zAxisPopupContent.py:
```

    def stopZMove(self):
        '''
        
        Send the imediate stop command
        
        '''
        print("z-axis Stopped")
        self.data.quick_queue.put("!")
        with self.data.gcode_queue.mutex:
            self.data.gcode_queue.queue.clear()

```

krkeegan, 2017-12-07:
That looks good.  I leave it up to you guys, but I would be fine if the font matched the stop command in the normal GC menu.  Because it spans two rows, it is already rather prominent.

blurfl, 2017-12-07:
To make sure I understand, the same size and color as the rest of the buttons, yes?
While we’re here, does it bother you that ‘raise’ is below ‘lower’? It feels upside down to me, but not enough to make an issue of 😊

BarbourSmith, 2017-12-07:
Yes! The ordering of raise and lower is going to drive me crazy now that you pointed it out! If you don't swap them I totally will 😀 

I vote same size and color as the rest of the buttons for now, but let's revisit that when we do the GUI overhaul next week. If we want I can have a nice red logo drawn up to match the home screen button

krkeegan, 2017-12-07:
Looks great. That was awesomely fast.

Closed by #491

closed 2017-12-07 by krkeegan