MaslowCNC/GroundControl #494: Auto Cal Step 10 Triangular version improvements

blurfl, 2017-12-14:
The instructions for this step would be better if they included the target dimension so the user can do a reality-check on a measurement. 
 Further, the software could do a bounds check anbd refuse values that are wildly wrong. Instead the wild value is written to groundcontrol.ini and the machine will be unable to find a valid position in the workarea.

BarbourSmith, 2017-12-14:
The distance is added in #495 

I think checking for reasonable values should be done in a pull request which removes the text box where the distance is entered and replaces it with a touch compatible key pad which could do the testing. We already have a keypad widget and adding a "reasonable value" check to it wouldn't be too difficult

blurfl, 2017-12-14:
Thanks for adding the text 👍. The pop up would be great, I messed around a bit but didn’t see a way to put up a message. 
While you’re in there, the text printed in the triangular routine refers to changing motor spacing instead of radius

krkeegan, 2017-12-16:
On a related note, there should be a sanity check for the dimension entered by the user.  For example, it shouldn't every calculate a negative radius, which I have seen it do

It should also capture if the user fails to change from MM - Inches.  We don't have to fix it for them, but we should at least give some sort of warning that the value looks very wrong.

BarbourSmith, 2017-12-18:
Yes! My plan for this was to:

1) Move the calibration process out into it's own object (because having everything in one file was ugly, and we want to be able to run the test pattern without going through the whole process some day)
2) Switch to using the on screen keypad widget to enter the cut distance making it so a keyboard is not required
3) Add a sanity check to the entered value

blurfl, 2017-12-22:
Wishes have been fulfilled!

closed 2017-12-22 by blurfl