MaslowCNC/GroundControl #500: Rotate 1 deg CW/CCW not really 1 deg

madgrizzle, 2017-12-18:
The code for turning a sprocket 1 deg actually turns the sprocket so as to feed out 0.5 mm of chain (B09 +/- .5)  Based upon a 10.1 sprocket radius, this would mean each command would advance the sprocket 2.84 degrees.  So depending upon where you end up in your alignment, you could conceivably be off by up to 1.42 degrees and not get any closer unless you swung around again (at least that's how I see it).

I don't know how big of an issue this actually is, but if it fed out 0.1763 mm of chain (B09 +/- 0.1763) then you would have a lot more precision in setting a tooth to 12 o'clock and it would be repeatable because 0.1763 mm  x 360 is fairly close to the actual circumference of the sprocket.. with the rounding of values you can't quite get there exactly.

If it matters, I suggest changing the value but adding buttons that do 5 degrees as well because pressing a button up to 180 times to align a tooth would be a little annoying.  Thoughts?  I could give a shot at coding this if it makes sense.

krkeegan, 2017-12-18:
A few ideas:

- We could do it like chain extension.  Offer CW and CCW 2 and 1 "degree" steps.  
- We could actually calculate the degree so that we attempt to move 1 degree, even if users have non-standard setups. (I think as part of the setup process we need to tell non-standard setups to plug in their data into the advanced tab first.)

blurfl, 2017-12-18:
How about a 5- or 10-degree option instead of 2-degrees, to get there with fewer clicks?

BarbourSmith, 2017-12-18:
My main goal here was to not confuse people. I tried to find wording that was easy and intuitive to understand and I felt like "one degree" seemed like a clear concept that clicking that button would move the sprocket a very small amount one way or the other. 

I vote for just changing the .5mm calculation so that it is actually one degree. If we do a true 1 degree with each click the most someone would have to click is 18 times to get to the nearest tooth...although to be honest can anyone get it to be vertical to 1 degree of accuracy anyway? Only having to click 6 times is nice 😀

madgrizzle, 2017-12-18:
The benefit of having additional buttons is if you have a particular tooth marked and you are trying to reset everything, you could get it to 12 o'clock a lot quicker with a 5 degree option as well.

blurfl, 2017-12-18:
I cut a small piece of refrigerator magnet to mark my king tooth, only need to get to the nearest one and move the marker. 😁

davidelang, 2017-12-18:
we should also add some fine tuning buttons that turn the motors less (down to 
individual encoder steps ideally)

limiting the adjustment to 0.5mm at it's finest is FAR too course

madgrizzle, 2017-12-21:
I'm working on a PR that will make 1 deg really 1 deg and add a 5 deg increment as well and want to test it be pushing it.  What I've done is to calculate the distance to move to gearTeeth and chainPitch from Advanced Settings, multiply them together and for 1 deg, divide by 360 and for 5 deg, divide by 72.  This way, if a different sprocket and/or chain is used, it automatically gets updated.  Based upon how the code is already laid out, this will be calculated every time you hit the button, but I can't imagine that the "get" command is really CPU intensive that this would be a problem.  On kivy, I made the row of buttons to be two columns, left column are the 1 degree increments and right column is the 5 degree increments.  The 'set zero' button is in the left column and googling about having kivy buttons spanning columns didn't come up with anything simple so I left it as it is.

I will test tonight and if it works, I'll submit the PR

davidelang, 2017-12-21:
Please add a finer movement as well (.1 degree or something like that) 
especially with larger gears 1 degree is not fine enough.

madgrizzle, 2018-01-04:
PR #542 has been submitted that provides 0.1, 1, and 5 degree increments.  I have not tested it on actual hardware as its too cold to work in the shed.

madgrizzle, 2018-01-05:
Going through the measureMachinePopup.py code, I found defs for LeftCW, LeftCCW, RightCW, RightCCW.  The setSprocketsVertical.py code also has defs for those.  Can they be deleted from measureMachinePopup.py since the only time they are used is for setting the sprockets vertical.  How does python handle definitions of the same functions?  Does the current code call setSprocketsVertical's functions or measureMachinePopup's functions.

BarbourSmith, 2018-01-05:
Good catch! Those were left over from before I split setSprocketsVertical into a seperate file. My bad for leaving them in there 😬

BarbourSmith, 2018-01-10:
I am going to close this as all taken care of!

closed 2018-01-10 by BarbourSmith