MaslowCNC/GroundControl #504: listSerialPorts slow with bluetooth enabled.

madgrizzle, 2017-12-21:
My laptop has some bluetooth serial ports enabled (not sure why, but COM3 and COM4 are bluetooth.. don't recall ever doing anything with bluetooth serial, nonetheless, they are there).  When I press the actions button, GC calls listSerialPorts and, with my Windows 10 computer, tries to open COM1 through COM256.  When it hits COM3 and COM4, it pauses for seconds... seemingly giving it time to establish a bluetooth serial connection.

It appears the purpose of opening then immediately closing the port is to determine which of the 256 comports exist (at least for Windows).  However, pyserial has serial.tools.list_ports.comports function that enumerates my COM3 and COM4 into a list.  For linux / osx, is there a particular need to test comports or do ya'll think we can just switch over to using serial.tools.list_ports.comports and not go through the open/close routine?

BarbourSmith, 2017-12-21:
Great deduction about what is going on! I have a vague recollection that the code you are talking about was added to fix and issue where serial.tools.list_ports was causing a bug...does anyone remember what the issue with using the function was?

blurfl, 2017-12-21:
How about opening a PR - I can test it on osx and one flavor of linux.

madgrizzle, 2017-12-21:
Found a stackoverflow thread that describes the issue (GC uses the precise code that was shown there).  However, the bug might have been fixed.

[https://stackoverflow.com/questions/12090503/listing-available-com-ports-with-python](url)

I'll work on a PR to test it.

blurfl, 2017-12-23:
It looks like serial.tools.list_ports.comports() does work, it builds a list with only the open ports.  I don't have a way to reproduce the slow response though (I'm on a VM, no BLE devices available).

blurfl, 2017-12-23:
There seem to be a number functions in connectMenu.py that are duplicated in serialPort.py but never used.
 It looks like updatePorts(), and detectCOMports() could be removed from serialPort.py, and the 'win' portion of listSerialPorts() moved into connectMenu.py.
 The 'linux/cygwin' and 'darwin' portions of connectMenu.updatePorts() are one-liners, the 'win' section needs the logic in listSerialPorts()
 Does it make sense to move these serial port discovery routines into connectMenu.py and leave serialPort.py with setPort(), openConnection() and connect()?

BarbourSmith, 2017-12-23:
>Does it make sense to move these serial port discovery routines into connectMenu.py and leave serialPort.py with setPort(), openConnection() and connect()?

Yes, I think that makes a lot of sense. This code is one of the parts which has been a carryover from the original tkinter based UI which means it's probably ripe for a clean up and reshuffle

blurfl, 2017-12-23:
@madgrizzle, I could hold off moving things around if you're nearly ready with a PR, or I could make the changes with the serial.tools.list_ports.comport() call. I don't want to get in your way here 😁

madgrizzle, 2017-12-23:
I haven't done a PR.  Been busy trying to complete (i.e., paint) a project for someone for Christmas.

blurfl, 2017-12-23:
That’s a good arrangement of priorities! I’ll put in a PR, then, but I don’t have a way to tell whether it takes care of this issue.

BarbourSmith, 2018-01-02:
What was the verdict? Did this fix it?

madgrizzle, 2018-01-03:
The problem is not fixed.  The code that is causing in issue is where each port is opened to "verify" that its an actual port.  Previously, the Windows code would set up a list of 256 possible COM ports and then would try to open each one to see which were valid.  The valids were retained.  The serial.tools.list_ports.comports() eliminates the need to do this for Windows as it returns the list of valid of COM ports.  However, I don't know what it does on linux/OSX.  It's possible on one or both of those OS's, that serial.tools.list_ports.comports() returns more than just valid ports.  If so, then you need to do the testing for those OS's.  I don't have linux or OSX setup to test this out.  We could modify the code to skip the testing portion for Windows or someone with linux and OSX can see what serial.tools.list_ports.comports() returns.

blurfl, 2018-01-03:
So this doesn't solve the problem under Windows? The PR special-cases Windows, doesn't change the behavior of Linux or OSX.
It sounds like the Bluetooth COM ports are slow to respond under Windows. Can you disable them in the Device manager? 
Is there a benefit to keeping this approach and only checking existing COM ports on Windows?

madgrizzle, 2018-01-03:
The only thing the PR does is, instead of creating a list of all 256 possible COM ports, it creates a list of existing comports.  This is good, but that wasn't what was causing the delay.  The delay occurs a little later in the code where it tries to open and then close each port.  The ones that do not return an error message are kept.  It's my position that you don't need to open and then close each port on Windows if you have a list from serial.tools.list_ports.comports().  I'll do a PR to illustrate.

davidelang, 2018-01-03:
trying to open every serial port is a bad idea, it means that if you have 
anything else using a serial port, you will interfere with it.

listing what ports exist on the system is fine, but trying to auto-detect which 
port a device is on should be a configurable option at most.

madgrizzle, 2018-01-03:
PR #540 [https://github.com/MaslowCNC/GroundControl/pull/540](url) has my proposed fix.  You could clean it up more by deleting the platform check in listSerialPorts, but I left it there in the event that this routine eventually gets used by linux/OSX.  One superfluous if statement in a UI isn't the end of the world.

madgrizzle, 2018-01-04:
Since I created this issue, I assume I have the ability to close it.  So I am hitting the "Close and comment" button and if it works, you'll see this message.

closed 2018-01-04 by madgrizzle