MaslowCNC/GroundControl #507: Double character in

blurfl, 2017-12-21:
PR#503 has introduced an issue where after changing the MM/INCHES in the number input characters typed are doubled.

BarbourSmith, 2017-12-21:
WEIRD! I would not have expected that. I will look into that right away! Thanks for the catch!

BarbourSmith, 2017-12-21:
I'm not seeing the same behavior, help me better understand what you are seeing

We're talking about the numbers typed in here showing up twice, right?

[image]

blurfl, 2017-12-21:
Right. I open the dialog, type a number, change mm to inches, backspace and type.

BarbourSmith, 2017-12-21:
I still can't make it happen 🤕

BarbourSmith, 2017-12-21:
Maybe it's platform dependant?

blurfl, 2017-12-21:
I've just replicated the issue on Win10 and a second instance of OS X. Seems to be related to clicking in the number field:
1. open the 'machine movement' change window
2. type '12'
3. click in the number field
4. type '34' - see '3344'

BarbourSmith, 2017-12-21:
Got it! I totally see it too now. I was missing the type in the window field step

BarbourSmith, 2017-12-21:
I have a proposed solution in #509

closed 2017-12-22 by blurfl